JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #97: Unable to disable the Keyboard LED on a Acer Predator PH315-55 running on Ubuntu 22.04

alok-38, 2023-05-31:
It is a dual boot system.I followed the exact steps after cloning the kernel module. Also tried booting to windows and fiddling with the software provided by Acer for controlling the Keyboard LED.
Rebooted to Llinux.

alok-38, 2023-05-31:
Update on what just happened.
After booting to Linux with the keyboard LED still enabled, I rebooted to the BIOS to find that Secure boot was already disabled whereas Fastboot was enabled.
I disabled Fastboot and went to windows to confirm if everything is okay. It turned out to be okay. Booted back to Linux and it turned out to be okay as well!
I am wondering if this has anything to do with disabling Fastboot?
I will monitor the situation for a few days and leave another note if it is a good idea to tweak this module to make it work flawlessly on this system.

alok-38, 2023-05-31:
Update: In order to ensure that the Keyboard LED stays disabled, I have to first boot to windows and then to Linux.

zetchz, 2023-07-19:
Hi alok. I just got this laptop and am running ubuntu 22.04 updated version too. 

I am not able to control the rgb lights of the keyboard with this package. Did I installed wrongly or are you unable to control the lights too?

alok-38, 2023-07-19:
Which Acer model are you referring to?

zetchz, 2023-07-22:
System Information
	Manufacturer: Acer
	Product Name: Predator PH315-55
Base Board Information
	Manufacturer: ADL
	Product Name: Evoque_ADH
	Version: V1.10

alok-38, 2023-07-24:
Please try to follow the exact same steps which I did. It may or may not work for this model.
I would have dove deeper if I had knowledge of drivers and the Linux kernel.