JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #98: When will the AN515-58 have Turbo support?

LarryTheMagicDragon, 2023-06-03:
I have an Acer Nitro 5 AN515-58 and I've been using this module. Frankly I've been at the edge of my seat waiting to see if it gets turbo support, or if the video on how to reverse engineer these things gets released.

JafarAkhondali, 2023-06-03:
I'm so sorry to hear that I didn't release the videos sooner, I got into several personal issues. But I promise to continue the videos as soon as possible. At least one video in the coming week

LarryTheMagicDragon, 2023-06-03:
Thanks! I'm looking forward to it.

LarryTheMagicDragon, 2023-06-16:
Any updates on those videos? If you can't make them anytime soon, do you have recommendations for alternative resources to learn from?

JafarAkhondali, 2023-06-17:
Sorry everyday a new problems gets into the life 😄  Actually the main reason I'm creating the videos is to have a resource to learn this stuff. You may try DotPeek to decompile, detect WMI methods in source codes and by intercpting requests, read acer-wmi.c file in Linux kernel, compare this project source code with old acer-wmi.c file of linux kernel, read lwn.net and ubuntu wiki resources about WMI, etc. I'll try my best to upload the new video by tomorrow.

LarryTheMagicDragon, 2023-06-19:
I understand how having issues in your personal life can derail these kinds of projects, believe me. I understand, So you shouldn't lose sleep over me or anything. If making a proper tutorial is too much right now, a video that's just you talking about the techniques you use in an informal sense could still be useful. A 30 minute rant about how you did it would still be quite fascinating

JafarAkhondali, 2023-06-19:
@LarryTheMagicDragon You are right, but I would like to keep some quality as well instead of random talking, so I'll just try to make balance between quality and productivity 😄

JafarAkhondali, 2023-06-23:
second video done finally
https://youtu.be/Dm60ZXl698c
closing it for now

closed 2023-06-23 by JafarAkhondali