JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #99: Can't control fan speed in a PH315-52 model

lestcape, 2023-06-11:
# Model: `sudo dmidecode | grep "Product Name" -B 2 -A 4`
System Information
	Manufacturer: Acer
	Product Name: Predator PH315-52
	Version: V1.12
	Serial Number: NHQ5MAA001914005872600
	UUID: 6d1cb23d-a862-44ed-9971-1daeee76cc2c
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: CFL
	Product Name: Covini_CFS
	Version: V1.12
	Serial Number: NHQ5MAA001913000232600
	Asset Tag: Type2 - Board Asset Tag
	Features:

## Count of keyboard RGB zones: 0

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1 

### RGB keyboard works?  No
### Turbo button turn on fans?  Yes
### Turbo button turn on LED? No
### Turbo button activates overclock? I don' t know

Debian GNU/Linux 12 (bookworm)
Debian 6.1.27-1 (2023-05-08) x86_64 GNU/Linux

Contrary to the description present in this repository, my model have not an rgb keyboard as probably other models  have (a variation?) and i only want that the fan speed works, but it don' t work. I don't want to overclock the CPU.  My interest is to control the temperature low. My laptop shutdown sometimes if the temperature is to high.

So the command that i want to execute sometimes is:
./facer_rgb.py -s 9

But that command do nothing in my case. The installation was as a systemd method and without any error.

sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.aml
[dsdt.zip](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/11714253/dsdt.zip)

Secure boot is off:
[image]

My specific model:
[image]

JafarAkhondali, 2023-06-17:
I suggest you to use ncdu program for now, as this repo doesn't have required functionality for now.

snowyoneill, 2023-06-20:
@lestcape Check out my [Predator Sense](https://github.com/snowyoneill/Linux-PredatorSense) clone for linux. You can control fan speeds there

lestcape, 2023-07-11:
@JafarAkhondali Thanks, but i know it have not support. The reason to open the issue was that.

@snowyoneill Thanks, i will give it a try, but i don't want to use pip. So, i will see if i can install it without to use pip. If i can i will test it with my hardware, otherwise not. I think should be good to have a flatpak, snap or an appimage of that at minimum. pip could be good for develop, but is not ok release an app to be installed that way... Also i see it have undervolt as a dependency and i currently used [intel-undervolt](https://github.com/kitsunyan/intel-undervolt), probably is a good idea to also support it.

Anyway thanks both for the suggestion and the work done.

lestcape, 2023-07-11:
@snowyoneill This is not available on newer kernels:
https://github.com/snowyoneill/Linux-PredatorSense/blob/master/ecwrite.py#L1

Please see:
https://bugzilla.redhat.com/show_bug.cgi?id=1943318

Or this proposed workaround, that use a kernel module to recover the funtionality:
https://github.com/YoyPa/isw/issues/265#issue-1419231642

So, your program don't work anymore as is. Have a good day.

snowyoneill, 2023-07-11:
@lestcape what distro are you running? You can enable it with `sudo modprobe ec_sys write_support=1`

snowyoneill, 2023-07-11:
@lestcape I have already read through those resources as I ran into that error when i tried to update Ubuntu to 22.04. Using the above commands worked to restore access to the EC. As for the other distros i cannot comment as i have not tested them.

lestcape, 2023-07-11:
Issue opened: https://github.com/snowyoneill/Linux-PredatorSense/issues/3