JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #101: Unable to install the package

Setu-Gupta, 2023-06-28:
# Model: Predator PH315-54

# Count of keyboard RGB zones: 4

I was able to install the package initially and I was also able to get it to work. Then I installed the SystemD service to test it out and the service also worked. Up till now, there were no issues. Then I tried to uninstall the package by running `sudo uninstall_service.sh` followed by `sudo uninstall.sh`.

After that I again tried to install the package by running `sudo ./install.sh`. At this point I faced the following error:
```
make -C /lib/modules/6.3.9-arch1-1/build M=/home/setu/PredatorSense modules
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
[ 1005.422389] Code: 5d c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 3b 7a 0d 00 f7 d8 64 89 01 48
[ 1005.422390] RSP: 002b:00007ffc78392d78 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[ 1005.422394] RAX: ffffffffffffffda RBX: 000055a3a83366f0 RCX: 00007f8d7d5192ed
[ 1005.422395] RDX: 0000000000000000 RSI: 000055a3a654ecb2 RDI: 0000000000000003
[ 1005.422396] RBP: 000055a3a654ecb2 R08: 0000000000000000 R09: 000055a3a8338ed0
[ 1005.422397] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000000000
[ 1005.422398] R13: 000055a3a8336360 R14: 00007ffc78392fb8 R15: 000055a3a8336800
[ 1005.422401]  </TASK>
[ 1005.422402] kobject_add_internal failed for acer-gkbbl with -EEXIST, don't try to register things with the same name in the same directory.
[ 1005.435610] input: Acer WMI hotkeys as /devices/virtual/input/input34
[*] Done

```

As you can see, the `acer-gkbbl` object already exists in the kernel. However, there is no character device available for it in `/dev`. I think that the uninstall script did not remove the `acer-gkbbl` object due to which I'm facing this error. How can I manually remove this device?

Setu-Gupta, 2023-06-28:
I somewhat fixed this issue by reinstalling the kernel. I tried to install the package once more and I got the following error:
```
$ sudo ./install.sh
make -C /lib/modules/6.3.9-arch1-1/build M=/home/setu/PredatorSense modules
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
[   71.581271] Code: 5d c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 3b 7a 0d 00 f7 d8 64 89 01 48
[   71.581272] RSP: 002b:00007ffc8955ea08 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[   71.581275] RAX: ffffffffffffffda RBX: 000055d94dadb6f0 RCX: 00007fc66a3192ed
[   71.581277] RDX: 0000000000000000 RSI: 000055d94d8fdcb2 RDI: 0000000000000003
[   71.581278] RBP: 000055d94d8fdcb2 R08: 0000000000000000 R09: 000055d94dadded0
[   71.581279] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000000000
[   71.581280] R13: 000055d94dadb360 R14: 00007ffc8955ec48 R15: 000055d94dadb800
[   71.581283]  </TASK>
[   71.581284] kobject_add_internal failed for acer-gkbbl with -EEXIST, don't try to register things with the same name in the same directory.
[   71.593599] input: Acer WMI hotkeys as /devices/virtual/input/input26
[*] Done
```

Setu-Gupta, 2023-06-29:
# Update

After a full system update and restarting my laptop multiple times, I was able to install the package again. However, this time, when I tried to change to keyboard lighting, my screen brightness went down to 0 and my laptop got hanged.
```
$ sudo ./install.sh
make -C /lib/modules/6.3.9-arch1-1/build M=/home/setu/PredatorSense modules
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
[    9.307553] JFS: nTxBlock = 8192, nTxLock = 65536
[   10.574656] systemd-journald[300]: File /var/log/journal/f5077fdbde22495eb2768a9a0e7d1625/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   10.807507] warning: `flameshot' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[   11.115285] Bluetooth: RFCOMM TTY layer initialized
[   11.115291] Bluetooth: RFCOMM socket layer initialized
[   11.115293] Bluetooth: RFCOMM ver 1.11
[   22.606572] acer_wmi: Acer Laptop WMI Extras unloaded
[   22.635272] facer: Acer Laptop ACPI-WMI Extras
[   22.635318] facer: Function bitmap for Communication Button: 0x801
[   22.647608] input: Acer WMI hotkeys as /devices/virtual/input/input24
[*] Done
```

After forcefully restarting my laptop, I tried to reinstall the package once more. This time, I did not face any issues with the screen brightness and my laptop also didn't hang. To figure out what happened the last time, I uninstalled the packed and tried to install it once more. On the second installation, however, I again faced the same error.
```
$ sudo ./install.sh
make -C /lib/modules/6.3.9-arch1-1/build M=/home/setu/PredatorSense modules
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
[    9.307553] JFS: nTxBlock = 8192, nTxLock = 65536
[   10.574656] systemd-journald[300]: File /var/log/journal/f5077fdbde22495eb2768a9a0e7d1625/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   10.807507] warning: `flameshot' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[   11.115285] Bluetooth: RFCOMM TTY layer initialized
[   11.115291] Bluetooth: RFCOMM socket layer initialized
[   11.115293] Bluetooth: RFCOMM ver 1.11
[   22.606572] acer_wmi: Acer Laptop WMI Extras unloaded
[   22.635272] facer: Acer Laptop ACPI-WMI Extras
[   22.635318] facer: Function bitmap for Communication Button: 0x801
[   22.647608] input: Acer WMI hotkeys as /devices/virtual/input/input24
[*] Done

$ sudo ./uninstall.sh    

$ sudo ./install.sh
make -C /lib/modules/6.3.9-arch1-1/build M=/home/setu/PredatorSense modules
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
[   39.839098] Code: 5d c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 3b 7a 0d 00 f7 d8 64 89 01 48
[   39.839099] RSP: 002b:00007ffe5222f688 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[   39.839102] RAX: ffffffffffffffda RBX: 000055b60267b6f0 RCX: 00007f56de7192ed
[   39.839104] RDX: 0000000000000000 RSI: 000055b600c7dcb2 RDI: 0000000000000003
[   39.839105] RBP: 000055b600c7dcb2 R08: 0000000000000000 R09: 000055b60267ded0
[   39.839107] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000000000
[   39.839108] R13: 000055b60267b360 R14: 00007ffe5222f8c8 R15: 000055b60267b800
[   39.839110]  </TASK>
[   39.839111] kobject_add_internal failed for acer-gkbbl with -EEXIST, don't try to register things with the same name in the same directory.
[   39.851593] input: Acer WMI hotkeys as /devices/virtual/input/input26
[*] Done
```

I suspect that there is a bug in the module exit code, due to which when the module is unloaded, the kernel object associated with the `acer-gkkbl` character device doesn't get removed.

JafarAkhondali, 2023-09-26:
Fixed in last update

closed 2023-09-26 by JafarAkhondali