JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #102: Installation issue

lasti15, 2023-06-30:
# Model: Predator PT314-52s

### Count of keyboard RGB zones: ?

### Count of Cpu Fans:  ?
### Count of Gpu Fans: ? 

### RGB keyboard works?  No
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? No

Hello - trying to install using:
`git clone https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module
cd "acer-predator-turbo-and-rgb-keyboard-linux-module"
chmod +x ./*.sh
sudo ./install.sh`

and getting this error:

```
laurence@Calayati:~/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo ./install.sh
make -C /lib/modules/5.19.0-46-generic/build M=/home/laurence/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-46-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-46-generic'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
rmmod: ERROR: Module acer_wmi is not currently loaded
insmod: ERROR: could not insert module src/facer.ko: Key was rejected by service
[ 6318.055409] audit: type=1400 audit(1688153088.791:206462): apparmor="DENIED" operation="ptrace" class="ptrace" profile="snap.discord.discord" pid=12579 comm="Utils" requested_mask="read" denied_mask="read" peer="unconfined"
[ 6318.055434] audit: type=1400 audit(1688153088.791:206463): apparmor="DENIED" operation="ptrace" class="ptrace" profile="snap.discord.discord" pid=12579 comm="Utils" requested_mask="read" denied_mask="read" peer="unconfined"
[ 6318.055447] audit: type=1400 audit(1688153088.791:206464): apparmor="DENIED" operation="ptrace" class="ptrace" profile="snap.discord.discord" pid=12579 comm="Utils" requested_mask="read" denied_mask="read" peer="unconfined"
[ 6318.055536] audit: type=1400 audit(1688153088.791:206465): apparmor="DENIED" operation="ptrace" class="ptrace" profile="snap.discord.discord" pid=12579 comm="Utils" requested_mask="read" denied_mask="read" peer="unconfined"
[ 6318.055560] audit: type=1400 audit(1688153088.791:206466): apparmor="DENIED" operation="ptrace" class="ptrace" profile="snap.discord.discord" pid=12579 comm="Utils" requested_mask="read" denied_mask="read" peer="unconfined"
[ 6318.055584] audit: type=1400 audit(1688153088.791:206467): apparmor="DENIED" operation="ptrace" class="ptrace" profile="snap.discord.discord" pid=12579 comm="Utils" requested_mask="read" denied_mask="read" peer="unconfined"
[ 6318.055607] audit: type=1400 audit(1688153088.791:206468): apparmor="DENIED" operation="ptrace" class="ptrace" profile="snap.discord.discord" pid=12579 comm="Utils" requested_mask="read" denied_mask="read" peer="unconfined"
[ 6318.055724] audit: type=1400 audit(1688153088.791:206469): apparmor="DENIED" operation="ptrace" class="ptrace" profile="snap.discord.discord" pid=12579 comm="Utils" requested_mask="read" denied_mask="read" peer="snap.snap-store.ubuntu-software-local-file"
[ 6318.055745] audit: type=1400 audit(1688153088.791:206470): apparmor="DENIED" operation="ptrace" class="ptrace" profile="snap.discord.discord" pid=12579 comm="Utils" requested_mask="read" denied_mask="read" peer="snap.snap-store.ubuntu-software-local-file"
[ 6323.697508] Loading of module with unavailable key is rejected
[*] Done
```

Following other issues, I've tried adding:
```
#include <linux/uaccess.h> 
#include <asm/uaccess.h> 
```
to src/facer.c to no avail. Any tips?

Cheers

Laurence

lasti15, 2023-07-03:
Commenting out rmmod acer_wmi in install.sh and installing` dkms and enrolling the default MOK stopped the error (hooray) but now it just hangs after:
```
make -C /lib/modules/5.19.0-46-generic/build M=/home/laurence/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-46-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-46-generic'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
```

lasti15, 2023-07-03:
install_service seems more promising, but now getting the turbofan error similar to https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/50

```
[Mode: install]
['turbo-fan' service is presented. Remove it.]
Synchronizing state of turbo-fan.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable turbo-fan
update-rc.d: error: turbo-fan Default-Start contains no runlevels, aborting.
[Remove old data]
removed '/opt/turbo-fan/install_openrc.sh'
removed '/opt/turbo-fan/facer_rgb.py'
removed '/opt/turbo-fan/keyboard.webp'
removed '/opt/turbo-fan/uninstall_service.sh'
removed '/opt/turbo-fan/Makefile'
removed '/opt/turbo-fan/LICENSE'
removed '/opt/turbo-fan/PKGBUILD/makepkg.sh'
removed '/opt/turbo-fan/PKGBUILD/PKGBUILD'
removed '/opt/turbo-fan/PKGBUILD/Predator-Sense-systemd-git.install'
removed directory '/opt/turbo-fan/PKGBUILD'
removed '/opt/turbo-fan/.Module.symvers.cmd'
removed '/opt/turbo-fan/src/.facer.ko.cmd'
removed '/opt/turbo-fan/src/facer.ko'
removed '/opt/turbo-fan/src/facer.mod.o'
removed '/opt/turbo-fan/src/facer.mod'
removed '/opt/turbo-fan/src/facer.mod.c'
removed '/opt/turbo-fan/src/.facer.o.cmd'
removed '/opt/turbo-fan/src/.facer.mod.o.cmd'
removed '/opt/turbo-fan/src/facer.o'
removed '/opt/turbo-fan/src/.facer.mod.cmd'
removed '/opt/turbo-fan/src/facer.c'
removed directory '/opt/turbo-fan/src'
removed '/opt/turbo-fan/modules.order'
removed '/opt/turbo-fan/install_service.sh'
removed '/opt/turbo-fan/Module.symvers'
removed '/opt/turbo-fan/.modules.order.cmd'
removed '/opt/turbo-fan/install.sh'
removed '/opt/turbo-fan/README.md'
removed '/opt/turbo-fan/refresh.sh'
removed '/opt/turbo-fan/uninstall.sh'
removed '/opt/turbo-fan/push.sh'
removed directory '/opt/turbo-fan'
[Create directories]
[Copy new data]
sending incremental file list
LICENSE
Makefile
Module.symvers
README.md
facer_rgb.py
install.sh
install_openrc.sh
install_service.sh
keyboard.webp
modules.order
push.sh
refresh.sh
uninstall.sh
uninstall_service.sh
PKGBUILD/
PKGBUILD/PKGBUILD
PKGBUILD/Predator-Sense-systemd-git.install
PKGBUILD/makepkg.sh
src/
src/.facer.ko.cmd
src/.facer.mod.cmd
src/.facer.mod.o.cmd
src/.facer.o.cmd
src/facer.c
src/facer.ko
src/facer.mod
src/facer.mod.c
src/facer.mod.o
src/facer.o

sent 3,790,101 bytes  received 545 bytes  7,581,292.00 bytes/sec
total size is 3,787,268  speedup is 1.00
[Create turbo-fan service]
Synchronizing state of turbo-fan.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable turbo-fan
update-rc.d: error: turbo-fan Default-Start contains no runlevels, aborting.
```

nxceurin, 2024-07-03:
For the turbo-fan error (after adding quirk), check where the service is with `systemctl status turbo-fan.service | grep "Loaded"`.
If it's in `/etc/init.d/turbo-fan`, try `sudo mv /etc/init.d/turbo-fan /etc/systemd/system/`

that seemed to solve the issue for me