JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #105: please add PH315-55

w74anheda, 2023-07-19:
this package not working in model PH315-55 
please support this model

JafarAkhondali, 2023-07-19:
How many fans do you have per CPU and GPU?

w74anheda, 2023-07-21:
how can find count of fans?

w74anheda, 2023-07-21:
after install this package : https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module
backlight is working but turbo fan button is not working

zetchz, 2023-07-22:
hi w74anheda. I have a PH315-55 too. can i check if you keyboard backlight can be controlled and whether the base board is the same? The backlight script doesnt  work for me.

---
 sudo dmidecode | grep "Product Name" -B 2 -A 4

System Information
	Manufacturer: Acer
	Product Name: Predator PH315-55

Base Board Information
	Manufacturer: ADL
	Product Name: Evoque_ADH
	Version: V1.10

w74anheda, 2023-07-23:
> sudo dmidecode | grep "Product Name" -B 2 -A 4

hi zetchz,
yes i can controll that with listed of command in package

System Information
	Manufacturer: Acer
	Product Name: Predator PH315-55
	Version: V1.07
--
Base Board Information
	Manufacturer: ADL
	Product Name: Evoque_ADH
	Version: V1.07

w74anheda, 2023-07-23:
> How many fans do you have per CPU and GPU?

how can find count of fans?

closed 2023-07-27 by JafarAkhondali

bharata108mklt, 2024-02-13:
I have the same model. All functions work. 
System Information
	Manufacturer: Acer
	Product Name: Predator PH315-55
	Version: V1.05

commit 4379a9ea (2023-07-27): [Add new models for turbo mode]
+ Add support for PT314-52S model - Fix #108
+ Add support for PT516-52s model - Fix #92
+ Add support for PH315-55  model - Fix #105