JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #106: Installs well and works fine but doesn't work after restart

Shawshank-dulal, 2023-07-21:
# Model: Predator PH315-54

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  2
### Count of Gpu Fans: 0  

### RGB keyboard works?  Yes
### Turbo button turn on fans?  Yes
### Turbo button turn on LED? Yes
### Turbo button activates overclock? No

The main Problem here is that it does not work after restart. Am I missing something?

JafarAkhondali, 2023-07-21:
Hey, Yes probably you are missing this https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module#install-as-a-systemd-service-will-work-after-reboot

Shawshank-dulal, 2023-07-21:
I had installed using that method

Shawshank-dulal, 2023-07-28:
Hey, I uninstalled and performed installation once again and now the rgb is working after reboot but turbo button is not.. what could be the reason?

JafarAkhondali, 2023-07-28:
That's weird ... If I was you, I would repeat the steps again, or also try using older commits