JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #109: Tried to install it but I get this error

futureHhacker, 2023-07-26:
kerol@kerol-Predator-PH317-53:~$ acer-predator-turbo-and-rgb-keyboard-linux-module/
bash: acer-predator-turbo-and-rgb-keyboard-linux-module/: Is a directory
kerol@kerol-Predator-PH317-53:~$ cd acer-predator-turbo-and-rgb-keyboard-linux-module/
kerol@kerol-Predator-PH317-53:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ chmod +x ./acer-predator-turbo-and-rgb-keyboard-linux-module.sh
chmod: cannot access './acer-predator-turbo-and-rgb-keyboard-linux-module.sh': No such file or directory
kerol@kerol-Predator-PH317-53:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ ls
facer_rgb.py        install.sh     Makefile  README.md   uninstall_service.sh
install_openrc.sh   keyboard.webp  PKGBUILD  refresh.sh  uninstall.sh
install_service.sh  LICENSE        push.sh   src
kerol@kerol-Predator-PH317-53:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ chmod +x ./install.sh
kerol@kerol-Predator-PH317-53:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo ./install_service.sh
[sudo] password for kerol: 
[Mode: install]
['turbo-fan' service is presented. Remove it.]
Synchronizing state of turbo-fan.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable turbo-fan
update-rc.d: error: turbo-fan Default-Start contains no runlevels, aborting.
[Remove old data]
removed '/opt/turbo-fan/install_openrc.sh'
removed '/opt/turbo-fan/Makefile'
removed '/opt/turbo-fan/src/facer.ko'
removed '/opt/turbo-fan/src/facer.mod.c'
removed '/opt/turbo-fan/src/facer.mod.o'
removed '/opt/turbo-fan/src/.facer.o.cmd'
removed '/opt/turbo-fan/src/.facer.mod.cmd'
removed '/opt/turbo-fan/src/.facer.ko.cmd'
removed '/opt/turbo-fan/src/facer.c'
removed '/opt/turbo-fan/src/.facer.mod.o.cmd'
removed '/opt/turbo-fan/src/facer.o'
removed '/opt/turbo-fan/src/facer.mod'
removed directory '/opt/turbo-fan/src'
removed '/opt/turbo-fan/install.sh'
removed '/opt/turbo-fan/refresh.sh'
removed '/opt/turbo-fan/PKGBUILD/PKGBUILD'
removed '/opt/turbo-fan/PKGBUILD/makepkg.sh'
removed '/opt/turbo-fan/PKGBUILD/Predator-Sense-systemd-git.install'
removed directory '/opt/turbo-fan/PKGBUILD'
removed '/opt/turbo-fan/facer_rgb.py'
removed '/opt/turbo-fan/LICENSE'
removed '/opt/turbo-fan/keyboard.webp'
removed '/opt/turbo-fan/modules.order'
removed '/opt/turbo-fan/README.md'
removed '/opt/turbo-fan/uninstall.sh'
removed '/opt/turbo-fan/.Module.symvers.cmd'
removed '/opt/turbo-fan/Module.symvers'
removed '/opt/turbo-fan/install_service.sh'
removed '/opt/turbo-fan/.modules.order.cmd'
removed '/opt/turbo-fan/push.sh'
removed '/opt/turbo-fan/uninstall_service.sh'
removed directory '/opt/turbo-fan'
[Create directories]
[Copy new data]
sending incremental file list
LICENSE
Makefile
README.md
facer_rgb.py
install.sh
install_openrc.sh
install_service.sh
keyboard.webp
push.sh
refresh.sh
uninstall.sh
uninstall_service.sh
PKGBUILD/
PKGBUILD/PKGBUILD
PKGBUILD/Predator-Sense-systemd-git.install
PKGBUILD/makepkg.sh
src/
src/facer.c

sent 2,651,857 bytes  received 336 bytes  5,304,386.00 bytes/sec
total size is 2,650,030  speedup is 1.00
[Create turbo-fan service]
Synchronizing state of turbo-fan.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable turbo-fan
update-rc.d: error: turbo-fan Default-Start contains no runlevels, aborting.
kerol@kerol-Predator-PH317-53:~/acer-predator-turbo-and-rgb-keyboard-linux-module$ ./facer_rgb.py
Traceback (most recent call last):
  File "/home/kerol/acer-predator-turbo-and-rgb-keyboard-linux-module/./facer_rgb.py", line 203, in <module>
    with open(CHARACTER_DEVICE, 'wb') as cd:
PermissionError: [Errno 13] Permission denied: '/dev/acer-gkbbl-0'
kerol@kerol-Predator-PH317-53:~/acer-predator-turbo-and-rgb-keyboard-linux-module$

JafarAkhondali, 2023-07-27:
Please format the logs properly :)
I believe running uninstall, reboot and installing as service should solve the issue as this model have been fully tested in this project

futureHhacker, 2023-08-03:
Dear @JafarAkhondali, tried it but I still get the same permission denied error.

JafarAkhondali, 2023-09-26:
Should be fixed with new update.