JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #110: Turbo mode not working Predator PT314-52s  ( and also not working RGB)

johandroid, 2023-07-29:
# Model: 
```
System Information
	Manufacturer: Acer
	Product Name: Predator PT314-52s
	Version: V1.07
	Serial Number: NHQFPEB001234081147600
	UUID: 9c5db92d-1e26-4d42-bf07-4d84db5bdd40
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: ADL
	Product Name: Cabrio_ADH
	Version: V1.07
	Serial Number: NBQFP11004234081147600
	Asset Tag: Type2 - Board Asset Tag
	Features:
```

## Count of keyboard RGB zones: 3

### Count of Cpu Fans:  2 (?) -> I'm not sure if the blades count as 1 or 2, or if there are more besides those blades
### Count of Gpu Fans: ? -> Not sure

### RGB keyboard works?  No -> When using the command `sudo ./facer_rgb.py -m 3 -s 5 -b 100` nothing happens.
### Turbo button turn on fans?  **No**
### Turbo button turn on LED? **No**
### Turbo button activates overclock? **No**

# System:
## Kernel
`uname -a`
```
Linux predator 6.4.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 27 Jul 2023 22:02:18 +0000 x86_64 GNU/Linux
```
## File system:
Encrypted BTRFS with LUKS, but I don't have a secure boot installed.

## Distro:
ArchLinux with GNOME
------------------- 
# Issue

## Installation
My main issue is related to the **fans on turbo mode**, my laptop is heating above 85º after few minutes, I can't work with it or will get **permanently damaged**.

When I run the command `sudo ./install_service.sh` I get this output:

```
[Mode: install]
[Remove old data]
[Create directories]
[Copy new data]
sending incremental file list
LICENSE
Makefile
Module.symvers
README.md
dmesg
facer_rgb.py
install.sh
install_openrc.sh
install_service.sh
keyboard.webp
modules.order
push.sh
refresh.sh
uninstall.sh
uninstall_service.sh
PKGBUILD/
[... 259 lines cut ...]
vivaldi_fmap           12288  1 atkbd
polyval_generic        12288  1 polyval_clmulni
gf128mul               16384  1 polyval_generic
ghash_clmulni_intel    16384  0
sha512_ssse3           45056  0
aesni_intel           360448  10
spi_intel_pci          12288  0
crypto_simd            16384  1 aesni_intel
cryptd                 28672  4 crypto_simd,ghash_clmulni_intel
xhci_pci               28672  0
spi_intel              36864  1 spi_intel_pci
i8042                  53248  1 facer
vmd                    28672  0
xhci_pci_renesas       24576  1 xhci_pci
serio                  28672  4 serio_raw,atkbd,i8042
btrfs                2035712  1
blake2b_generic        24576  0
libcrc32c              12288  1 btrfs
crc32c_generic         12288  0
crc32c_intel           16384  2
xor                    20480  1 btrfs
raid6_pq              122880  1 btrfs
```

@JafarAkhondali  thank you for your help, I hope you can guide me, I don't know what else to do, I this doesn't work I will not be able to use Linux on my Laptop since it will burn ( after some use it gets above 85º), I don't care about RGB, I'm interested in don't burn my laptop.

JafarAkhondali, 2023-07-30:
Hello. Can you try a live ubuntu to check if it works? If that didn't work, try this project instead: https://github.com/hirschmann/nbfc/

johandroid, 2023-07-30:
I think it could be difficult to try inside a Live Ubuntu since it's necessary to download a lot of staff to compile the kernel module, but I will try.

Regarding nbfc I have already tried to make it work.

I have been watching your videos (very nice videos btw, waiting for the remaining episodes), did you used reverse engineering for this laptop model? Or how did you get the registers to modify?

JafarAkhondali, 2023-08-11:
My laptop is a little different but the path should be very similar. I've uploaded the next video, https://www.youtube.com/watch?v=Cfjy-Amru-k so you can watch it. I used WMI instead of playing with registers directly as it was the same method for Acer.

JafarAkhondali, 2023-08-15:
Also I think checking it with live Ubuntu shouldn't be hard as you don't need to compile the kernel itself, it'll just need to compile this module which should be pretty fast.

geevarghesest, 2023-08-16:
Hi,
I tried this module with ubuntu 23.04 running 6.2 kernel on PT314-52s. Only RGB controls in static mode seems to work by specifying zones, this device has 3 zone RGB. The turbo button doesn't seem to be working but in README this model has turbo button tested as working, was this tested on a newer kernel version?

JafarAkhondali, 2023-10-07:
Can you check the issue again with latest version?  Uninstall - reboot - reinstall

geevarghesest, 2023-10-08:
> Can you check the issue again with latest version?  Uninstall - reboot - reinstall

Awesome! Turbo button and RGB dynamic effects work fine now. Thank you @IllusionMan1212 @JafarAkhondali

geevarghesest, 2023-10-08:
Just a related question, do you have any idea on why screen brightness changes along with keyboard when i use the function key for kb backlight and the turbo button light not turning on?

JafarAkhondali, 2023-10-08:
@geevarghesest This is a known bug but the cause is not know 😄 It should be possible to patch this behavior by modifying this driver ( basic idea: waiting for fb keypress, then keep brightness at the same level for 2 seconds in a loop).