JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #111: Kernel panic on 6.5 and newer kernel

jyash8, 2023-07-30:
Model: `Nitro AN515-58`

module is sometimes causing kernel panic

mainly observed when i shutdown my pc after after not pressing backlight brightness change buttons

mainly in kernel >= 6.5

jyash8, 2023-08-03:
so how much time could it take

JafarAkhondali, 2023-08-03:
@jyash8 Right now I barely can find time to record videos for RE :( I hope someone can work on this

jyash8, 2023-09-08:
I think i have maybe figured out the problem due to some reason facer module can not be removed.
I have attached the photo of the log when system panics.
[image]
Because when i tried to remove the facer module with modprobe it said "facer not found".

closed 2023-09-28 by jyash8