JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #112: Installation problem on PHN16-71

Sa2908, 2023-07-30:
I get the following error on running `sudo ./install.sh`

```
make -C /lib/modules/6.1.0-10-amd64/build M=/home/sanket/Desktop/Code/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-10-amd64'
  CC [M]  /home/sanket/Desktop/Code/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
  MODPOST /home/sanket/Desktop/Code/acer-predator-turbo-and-rgb-keyboard-linux-module/Module.symvers
  LD [M]  /home/sanket/Desktop/Code/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
  BTF [M] /home/sanket/Desktop/Code/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
Skipping BTF generation for /home/sanket/Desktop/Code/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-10-amd64'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
        sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
rmmod: ERROR: Module acer_wmi is not currently loaded
insmod: ERROR: could not insert module src/facer.ko: Key was rejected by service
[ 1867.639080] BPF:  
[ 1867.639082] BPF: Invalid name
[ 1867.639084] BPF: 
[ 1867.639087] failed to validate module [acer_wmi] BTF: -22
[ 1867.639221] acer_wmi: Acer Laptop ACPI-WMI Extras
[ 1867.639277] acer_wmi: Function bitmap for Communication Button: 0x801
[ 1867.653205] input: Acer WMI hotkeys as /devices/virtual/input/input23
[ 1879.602759] acer_wmi: Acer Laptop WMI Extras unloaded
[ 1879.635911] Loading of unsigned module is rejected
[ 2097.392544] Loading of unsigned module is rejected
[*] Done
````

Sa2908, 2023-07-30:
i am using debain 12 currently

Spideynati, 2023-08-07:
did you find any solution yet im having same issues.
my model is also phn16-71

Spideynati, 2023-08-16:
yes i disabled secureboot

JafarAkhondali, 2023-10-02:
I believe you didn't disable secure boot. The error msg clearly shows a problem of loading unsigned module.

Ref:
https://forums.centos.org/viewtopic.php?t=47224

Or maybe with help of this PR you can check what went wrong in case you want to use it with secure boot on:
https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/pull/96/files

Sa2908, 2023-11-05:
Yes thanks it works fine now

closed 2023-11-05 by JafarAkhondali

mitanshu7, 2024-05-08:
@JafarAkhondali, I could see the RGB lights under the keyboard working, however the fan speed did not increase when i pressed the button. Any solution?

Thanks in advance!

patrickalima98, 2024-05-08:
Hi @mitanshu7, I have this same model, and for me the keyboard RGB is not working, but the Turbo mode fan speed is working. I'm using PopOS 22.04, what do you do to work with RGB?

mitanshu7, 2024-05-08:
Hello, @patrickalima98. I am using Debian 12. I did the following:

1. ` sudo apt-get install linux-headers-$(uname -r) gcc make`
2. 
```
git clone https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module
cd "acer-predator-turbo-and-rgb-keyboard-linux-module"
chmod +x ./*.sh
sudo ./install.sh
```

3. Then I simply ran: `python facer_rgb.py` and the lights started to move in a wave like fashion from left to right.

However, the turbo button did not work. I have added a PR for the same.

patrickalima98, 2024-05-08:
Hi @mitanshu7, your model is the Helios Neo, right? So if yes, your keyboard has 4 zones. My model is the Helios 16 (ph16-71) and the RGB keyboard is per key. I think this is the problem, but thanks for your answer, I will try to make it work

mitanshu7, 2024-05-08:
Hi @patrickalima98, yes. My model is Helios Neo. 

#133 Added support for turbo mode for your model, however, per key rgb might not work. Checkout #13 as mentioned in #130!

Spideynati, 2024-05-09:
hey @mitanshu7 i have the same model and the turbo mode is not working for me either keyboard works though have you found a workaround yet?? (for turbo i.e.)

mitanshu7, 2024-05-09:
Hello @Spideynati, yes. My pull request has been merged in this repo. Please update the local repository and try again. After installing, press the turbo button and you should hear the fans coming up loudly.

Spideynati, 2024-05-09:
yeah its working now just saw the new pull request, btw does it automatically controls your fan to your workload or only blasts it in full speed in turbo mode (wierd but wanted to know), thanks btw @mitanshu7

mitanshu7, 2024-05-09:
Hi @Spideynati, sadly unlike windows, this one just blasts it at full speed. I believe it's still better than thermal throttling while running heavy workloads!

Spideynati, 2024-05-09:
yeah @mitanshu7, thats what i thought yeah but its still better than thermal throttling and the rgb works too.