JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #113: module stopped working after an update

Hitoshura12, 2023-08-05:
# Model: Predator PH315-55

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  0 
### Count of Gpu Fans: 2 ? at least that's what I think 

### RGB keyboard works?  Yes\No No, used to work but stopped working after an update.
### Turbo button turn on fans?  Yes\No No, Not concerned about turbo mode. RGB is what I want it working again.
### Turbo button turn on LED? Yes\No No
### Turbo button activates overclock? Yes\No No
System used to work fine for a few months, but after a recent update  (3 days ago), it suddenly stopped. Dual boot with Win11/PopOS.  I tried everything, also checked in other issues, nothing.

Hitoshura12, 2023-08-05:
additional info:
when I try to uninstall , this error pops up : 
`sudo ./uninstall.sh`
rmmod: ERROR: Module facer is not currently loaded
Update: 
After tryting to install without it being any service (normal one, without reboot), it is working, but I suspect it'll revert after rebooting.

Megamichi123, 2024-12-13:
Hello i have now the same problem on my Predator PHN18-71
and i tried to recompile it with "make":
```

$ make
make -C /lib/modules/6.8.0-50-generic/build M=/home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Verzeichnis „/usr/src/linux-headers-6.8.0-50-generic“ wird betreten
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  CC [M]  /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘acer_wmi_input_setup’:
/home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:3205:49: error: passing argument 2 of ‘wmi_install_notify_handler’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 3205 |                                                 acer_wmi_notify, NULL);
      |                                                 ^~~~~~~~~~~~~~~
      |                                                 |
      |                                                 void (*)(union acpi_object *, void *)
In file included from ./include/linux/i2c.h:13,
                 from ./include/uapi/linux/fb.h:6,
                 from ./include/linux/fb.h:7,
                 from /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:22:
./include/linux/acpi.h:401:60: note: expected ‘wmi_notify_handler’ {aka ‘void (*)(unsigned int,  void *)’} but argument is of type ‘void (*)(union acpi_object *, void *)’
  401 |                                         wmi_notify_handler handler, void *data);
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~
/home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: At top level:
/home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1977:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
 1977 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
/home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:3663:1: fatal error: opening dependency file /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/.facer.o.d: Keine Berechtigung
 3663 | module_exit(acer_wmi_exit);
      | ^~~~~~~~~~~
cc1: some warnings being treated as errors
compilation terminated.
make[3]: *** [scripts/Makefile.build:243: /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Fehler 1
make[2]: *** [/usr/src/linux-headers-6.8.0-50-generic/Makefile:1925: /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module] Fehler 2
make[1]: *** [Makefile:240: __sub-make] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-6.8.0-50-generic“ wird verlassen
make: *** [Makefile:14: default] Fehler 2

```

LarryTheMagicDragon, 2024-12-13:
@Megamichi123 Your error messages show that you have fewer lines in facer.c than the latest version. Have you tried downloading it again and seeing if it works?

Megamichi123, 2024-12-14:
it doesn't work even after re-downloading

LarryTheMagicDragon, 2024-12-14:
@Megamichi123 Can you show the output from running make, install.sh, or install_service.sh after re-downloading the latest version? The messages that show up when it fails usually show what needs to be done to fix it, if you know how to read them, and I can't really do much if I don't know what is happening.

Megamichi123, 2024-12-15:
1. uninstall:  
```
$ sudo ./uninstall_service.sh 
[Mode: remove]
['turbo-fan' service is presented. Remove it.]
Removed "/etc/systemd/system/multi-user.target.wants/turbo-fan.service".
[Remove old data]
removed '/opt/turbo-fan/install.sh'
removed '/opt/turbo-fan/install_openrc.sh'
removed '/opt/turbo-fan/service.sh'
removed '/opt/turbo-fan/modules.order'
removed '/opt/turbo-fan/facer_rgb.py'
removed '/opt/turbo-fan/src/facer.mod'
removed '/opt/turbo-fan/src/facer.c'
removed '/opt/turbo-fan/src/facer.mod.o'
removed '/opt/turbo-fan/src/.facer.ko.cmd'
removed '/opt/turbo-fan/src/facer.mod.c'
removed '/opt/turbo-fan/src/facer.o'
removed '/opt/turbo-fan/src/.facer.o.cmd'
removed '/opt/turbo-fan/src/.facer.mod.cmd'
removed '/opt/turbo-fan/src/facer.ko'
removed '/opt/turbo-fan/src/.facer.mod.o.cmd'
removed directory '/opt/turbo-fan/src'
removed '/opt/turbo-fan/push.sh'
removed '/opt/turbo-fan/dkms.conf'
removed '/opt/turbo-fan/keyboard.py'
removed '/opt/turbo-fan/refresh.sh'
removed '/opt/turbo-fan/uninstall.sh'
removed '/opt/turbo-fan/PKGBUILD/makepkg.sh'
removed '/opt/turbo-fan/PKGBUILD/PKGBUILD'
removed '/opt/turbo-fan/PKGBUILD/Predator-Sense-systemd-git.install'
removed directory '/opt/turbo-fan/PKGBUILD'
removed '/opt/turbo-fan/Module.symvers'
removed '/opt/turbo-fan/uninstall_service.sh'
removed '/opt/turbo-fan/README.md'
removed '/opt/turbo-fan/install_service.sh'
removed '/opt/turbo-fan/LICENSE'
removed '/opt/turbo-fan/Makefile'
removed '/opt/turbo-fan/keyboard.webp'
removed directory '/opt/turbo-fan'
``` 
2. reclone the repo
3. chmod
4. make
```
$ make
make -C /lib/modules/6.8.0-50-generic/build M=/home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-50-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  CC [M]  /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1998:20: warning: 'WMID_gaming_get_u64' defined but not used [-Wunused-function]
 1998 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
  MODPOST /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/Module.symvers
  CC [M]  /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.mod.o
  LD [M]  /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
  BTF [M] /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
Skipping BTF generation for /home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-50-generic'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi

``` 
5. install.sh
```
$ sudo ./install.sh
make -C /lib/modules/6.8.0-50-generic/build M=/home/megamichi/acer-predator-turbo-and-rgb-keyboard-linux-module modules
[... 49 lines cut ...]
uninstall.sh
uninstall_service.sh
PKGBUILD/
PKGBUILD/PKGBUILD
PKGBUILD/Predator-Sense-systemd-git.install
PKGBUILD/makepkg.sh
src/
src/.facer.ko.cmd
src/.facer.mod.cmd
src/.facer.mod.o.cmd
src/.facer.o.cmd
src/facer.c
src/facer.ko
src/facer.mod
src/facer.mod.c
src/facer.mod.o
src/facer.o

sent 4,172,846 bytes  received 583 bytes  8,346,858.00 bytes/sec
total size is 4,169,788  speedup is 1.00
[Create turbo-fan service]
chmod: cannot access '/opt/turbo-fan/service.sh': No such file or directory
Created symlink /etc/systemd/system/multi-user.target.wants/turbo-fan.service → /etc/systemd/system/turbo-fan.service.
``` 
> Treiber funktioniert immer noch nicht

LarryTheMagicDragon, 2024-12-15:
install.sh
`insmod: ERROR: could not insert module src/facer.ko: Key was rejected by service`
`[ 1748.999277] Loading of module with unavailable key is rejected`
as far as I understand it these lines indicate that there is an issue with secure boot. Do you have secure boot enabled?

Megamichi123, 2024-12-15:
yes I just turned it off and the driver works again, thank you 🥰