JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #115: Almost working on an515-58, the fan on the left doesnt work

p-aaditya, 2023-08-28:
# Model: Nitro AN515-58

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1

### RGB keyboard works?  Yes
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? Yes\No

since mine is a nitro i dont have a turbo button but i do have a nitrosense button so are they same?
next rgb backlight works so good
one of the fans seem to be working which is the one on the right when i run install so nice again
the other fan on the left dont seem to be working when i run install
so please help me in making the left fan work
thanks in advance

p-aaditya, 2023-08-29:
isnt your playlist more towards making software communicate with rgb lights but for me the rgb lights work fine except for the gpu fan not working so how should i proceed in order to solve this problem

JafarAkhondali, 2023-08-29:
Yes but they are similar. You can De-compile NitroSense and find out how the turbo mode function works.

jyash8, 2023-09-07:
I have the same model. The only workaround i know is to use nbfc-linux. Just pick config of any acer nitro close to the model. I had picked this config "Acer Nitro AN515-51". It is working fine and i am able change my fan manually and auto mode is also working fine.

p-aaditya, 2023-10-10:
> I have the same model. The only workaround i know is to use nbfc-linux. Just pick config of any acer nitro close to the model. I had picked this config "Acer Nitro AN515-51". It is working fine and i am able change my fan manually and auto mode is also working fine.

well i did try using nbfc-linux but that did not help. an515-51 config file would never change my current fan speed

p-aaditya, 2023-10-10:
or if i can get a config file that can help ass well

> I have the same model. The only workaround i know is to use nbfc-linux. Just pick config of any acer nitro close to the model. I had picked this config "Acer Nitro AN515-51". It is working fine and i am able change my fan manually and auto mode is also working fine.

jyash8, 2023-10-10:
Can you tell me the commands you used to try to change the fan speed

> I have the same model. The only workaround i know is to use nbfc-linux. Just pick config of any acer nitro close to the model. I had picked this config "Acer Nitro AN515-51". It is working fine and i am able change my fan manually and auto mode is also working fine.

jyash8, 2023-10-10:
Try this command
```shell
 sudo nbfc set -f 1 -a
```

p-aaditya, 2023-10-11:
> Try this command
> 
> ```shell
>  sudo nbfc set -f 1 -s 100
> ```
> 
> Your fan should ramp up to full speed.

i have used this command it only changes the target fan speed but the current fan speed in unaffected