JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #116: Not working on PH18-71

zChris83, 2023-09-02:
# Model:
System Information
	Manufacturer: Acer
	Product Name: Predator PH18-71
	Version: V1.03
	Serial Number: NHQKSED0073100589C7600
	UUID: 24558df0-683a-4b4a-9b0f-c01850da9527
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: RPL
	Product Name: EQS_RTX
	Version: V1.03
	Serial Number: NBQKS110023100589C7600
	Asset Tag: Type2 - Board Asset Tag
	Features:

## Count of keyboard RGB zones: 
I believe its 1 zone per button on the keyboard 

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1  

### RGB keyboard works?  No
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? Dont know but probably not

neilm-fourjs, 2023-09-08:
Same here:
System Information
        Manufacturer: Acer
        Product Name: Predator PH18-71
        Version: V1.03
        Serial Number: NHQKTEK0063121585C7600
        UUID: 661db047-de40-4141-94e3-c01850fc004a
        Wake-up Type: Power Switch
--
Base Board Information
        Manufacturer: RPL
        Product Name: EQS_RTX
        Version: V1.03
        Serial Number: NBQKT110013121585C7600
        Asset Tag: Type2 - Board Asset Tag
        Features:

RGB keyboard worked in Windows - before I wiped it and installed Kubuntu 22.04 LTS - kernel module builds and installs, device is created, but ./facer_rgb.py is doing nothing, tried lots of the settings.
$ l /sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56
lrwxrwxrwx 1 root root 0 Sep  8 17:40 /sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56 -> ../../../devices/pci0000:00/PNP0C14:00/wmi_bus/wmi_bus-PNP0C14:00/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56

[Fri Sep  8 17:44:36 2023] facer: Acer Laptop ACPI-WMI Extras
[Fri Sep  8 17:44:36 2023] facer: Function bitmap for Communication Button: 0x801
[Fri Sep  8 17:44:36 2023] input: Acer WMI hotkeys as /devices/virtual/input/input33

neilm-fourjs, 2023-09-08:
dsdt.aml zipped
[dsdt.zip](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/12562140/dsdt.zip)

zChris83, 2023-09-19:
Whats that ?

neilm-fourjs, 2023-09-19:
Checkout the Feedback section of the README.md
ie:
If this worked or didn't worked for you, kindly make a new issue, and attach the following if possible:
sudo dmidecode | grep "Product Name" -B 2 -A 4
sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.aml