JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #117: Predator PH315: turbo button

w74anheda, 2023-09-07:
after install as service, push turbo button and turbo mode is activate but after push that turbo mode is not deactivate and not working, restart my laptop but not resetting turbo mode

amirkfir, 2023-09-11:
I experience the same problem at ph315-55, turbo buttom turning fans on, but not off, and it's stuck in on mode after reboot.

w74anheda, 2023-09-15:
What should we do to fix it?

JafarAkhondali, 2023-09-25:
Right now turbo mode checks for the state of the turbo LED in to see if turbo mode is activated. Temporary you can disable it in windows, but perhaps the WMI call to check turbo LED state is wrong and should be fixed.

JafarAkhondali, 2023-10-07:
Can you check the issue again with latest version?  Uninstall - reboot - reinstall

w74anheda, 2023-10-09:
> Can you check the issue again with latest version? Uninstall - reboot - reinstall

yes its, work. thanks for that.

closed 2023-10-09 by JafarAkhondali