JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #118: dynamic backlight does not work

FauxJD, 2023-09-09:
# Model: NITRO AN515-46

## Count of keyboard RGB zones: 1

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1 

### RGB keyboard works?  Yes

FauxJD, 2023-09-09:
only this command "./facer_rgb.py -m 0 -z 4 -cR 255 -cB 255 -cG 0" works fine

JafarAkhondali, 2023-09-25:
Can you uninstall & install again the project?

FauxJD, 2023-09-26:
I'm trying this now. But the dynamic mode did not work (only changing the static mode works). And I didn't tell you that I have an LED indicator on the F8 button. Could we be talking about another revision of the keyboard?

JafarAkhondali, 2023-09-26:
Yes maybe your keyboard is completely different, i thought the recent update might fix your issue

JafarAkhondali, 2024-01-24:
Thanks for the update. Did turbo mode work for you?

FauxJD, 2024-01-24:
My keyboard doesnt have a turbo-boost button (if we are speak about fans), only have "nitro sence" button, indicators of microphone (It can normal works only in win11, in linux it all time glowing, but microphone is works fine ) and caps-lock.