JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #122: Issues on PH315-55

charmbyte, 2023-09-30:
First, thanks for your great work. However, I have two issues with PH315-55.

#1: The RGB does not seem to work. I've checked for the directory mentioned in the directions and it exists. I have all requirements install. Then I ran the install script with no issue. Finally, I tried running the python script but nothing happens. There's no error messages or anything, but also no RGB on the keyboard. I tried copying and pasting every example and none of them work.

#2: I've seen this mentioned before, but I thought I would echo it here since I'm sure it's at least useful information for you. The turbo button works when pressed but it does not turn off even after reboot. The only way to turn it off seems to be to run this uninstall script then rebooting.

anupp, 2023-09-30:
I can confirm for PH315-55 , the RGB and FANS(Turbo) do not work  
For the following:

System Information
	Manufacturer: Acer
	Product Name: Predator PH315-55
	Version: V1.08
	Serial Number: NHQGMEG0012231AE9B3400
	UUID: 0d898c27-7bd9-ec11-80e5-088fc36134a1
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: ADL
	Product Name: Evoque_ADH
	Version: V1.08
	Serial Number: NBQGM11003223270FA3400
	Asset Tag: Type2 - Board Asset Tag
	Features:

[dsdt.zip](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/files/12774913/dsdt.zip)

JafarAkhondali, 2023-10-01:
Sorry but for now I can't help much with it ... your best shot is to find it out yourself. Perhaps the videos I've created on YT can be a good starting point

charmbyte, 2023-10-01:
Ok, thank you. You may wish to update the readme. Currently it says RGB is supported and tested for PH315-55

asif-ansari, 2023-10-03:
1. RGB lights not working at all
2. Turbo mode does work without lights, and then you can't turn it off. Boot into windows or uninstall the module to turn off turbo mode

Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.8
KDE Frameworks Version: 5.110.0
Qt Version: 5.15.10
Kernel Version: 6.6.0-1-MANJARO (64-bit)
Graphics Platform: Wayland
Processors: 20 × 12th Gen Intel® Core™ i7-12700H
Memory: 62.5 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: Acer
Product Name: Predator PH315-55
System Version: V1.11

LechTheMitch, 2023-10-05:
Hey guys so is keyboard backlighting completely off for or does it have this weird pattern that can't be turned off 
because i really want to disable backlighting on the keyboard so if there is anyway to it other than this script please tell me because it doesn't work on our device as previously stated by other people in this issue

JafarAkhondali, 2023-10-07:
Can you check the issue again with latest version?  Uninstall - reboot - reinstall

LechTheMitch, 2023-10-11:
> Can you check the issue again with latest version? Uninstall - reboot - reinstall

Thank you turbo mode works fine know rgb still broken though appreciate your efforts :)

Pointing8422, 2023-11-07:
I can confirm the RGB isn't working, but the turbo is working fine.