JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #124: Improve installation process

JafarAkhondali, 2023-10-02:
Right now the installation process is a little manual. We can make it interactive (or with some CLI options) to detect service manager (systemd or openrc). It can also install requirements as well. Furthermore, we can check secure boot status and prevent installation if secure boot is enabled.

mmsaeed509, 2023-10-02:
I built a package for this module (for Arch Linux only) which is called `Predator-Sense-CLI`

and there is a Predator edition of Exodia OS that is customized for Acer Predator Laptops and comes with `Predator-Sense-CLI` pre-installed

for those who use Arch Linux, you can set up Exodia OS Repos to get updates for this module
- edit `/etc/pacman.conf`
- add Exodia Repo
 
  - ```
    ## Core packages repo ##
    [exodia-repo]
    SigLevel = Optional TrustAll
    Server = https://exodia-os.github.io/$repo/$arch
    ```
- sync the reps `sudo pacman -Syyy`
- install the module `sudo pacman -S Predator-Sense-CLI`
- you can install GUI app `sudo pacman -S Predator-Sense-GUI`

now you will get all updates automatically

JafarAkhondali, 2023-10-09:
Thanks @mmsaeed509 
We do also need to do something for non-arch users as well.