JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #127: Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000009

TangerineGameDev, 2023-10-24:
# Model:
```System Information
	Manufacturer: Acer
	Product Name: Nitro AN517-52
	Version: V2.08
	Serial Number: NHQAWEV00411118A0F3400
	UUID: d8d2661f-8188-eb11-80dc-089798e905b9
	Wake-up Type: Power Switch

Base Board Information
	Manufacturer: CML
	Product Name: Karoq_CMS
	Version: V2.08
	Serial Number: NBQB211002111D92173400
	Asset Tag: Type2 - Board Asset Tag
	Features:
```

Count of keyboard RGB zones: 4
Count of Cpu Fans:  1
Count of Gpu Fans: 1 

RGB keyboard works?  Yes
Turbo button turn on fans?  No
Turbo button turn on LED? No
Turbo button activates overclock? No

# Issue
Whenever shutting down my computer, I have a chance of panicking the kernel, with the error 0x00000009
whenever the facer module was the last unloaded module.

Img:
[image]

# Note
This model is unsupported! (Acer Nitro AN517-52)
Still the keyboard works! If you decide not to fix the issue, because it isn't officially supported here, I will be understanding! It's fine! (Open-source is hard lol)

JafarAkhondali, 2023-10-24:
Well, I have no idea about the reason of this issue. But another important issue is that the code in this project is based on the older acer-wmi.c file from linux kernel, and we are outdated now. Maybe latest acer-wmi.c won't have such issues but we'll need to implement missing parts into it, which requires some time tbh. It's not the hard yet we need someone to do it (and maybe send the new changes to linux kernel also.)

Eclextic, 2025-05-06:
@JafarAkhondali I haven't been following this repo in the last years. What has changed? Have we rewritten those drivers?

LarryTheMagicDragon, 2025-05-06:
@Eclextic @JafarAkhondali If I recall this modules code correctly, it calls some functions that are in other modules. using lsmod I can see that it depends on the modules platform_profile, sparse_keymap, rfkill, i8042, video, and wmi. If it's calling functions in modules that are unloaded that would cause the panic. I'll look through the module exit functions later to see if I can find any clues. It may also have something to do with there being no quirk_entry for AN517-52. When I look around I'll see about adding an entry for it.