JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #136: Building it for NixOS?

AligatorRicardo, 2023-12-22:
# Model: Nitro AN515-46
Firstly, this kernel module worked very well on my device. It changed my keyboard from plain red RGB to the same pattern used on Windows that was fine for me when using arch.
However, after I switched to NixOS I noticed that there is no way to build it since NixOS is immutable and is not FHS compliant. Does anyone know if it is possible to build it as a nixpkg?

AligatorRicardo, 2024-01-12:
Ok, a good update for those who want to use NixOS and want to use this module. I managed to package this, so here is the code.
First, write this acer-module.nix:
```

{ stdenv, lib, fetchFromGitHub, kernel, kmod }:
stdenv.mkDerivation rec {
  name = " acer-predator-turbo-and-rgb-keyboard-linux-module-${version}-${kernel.modDirVersion}";
  version = "main";

  src = fetchFromGitHub {
    owner = "JafarAkhondali";
    repo = "acer-predator-turbo-and-rgb-keyboard-linux-module";
    rev = "v${version}";
    sha256 = "za8dbO5ZBaqdwohTgstyC3jbqOxLIq8bJuy0ASHWYew=";
  };

   setSourceRoot = ''
    export sourceRoot=$(pwd)/source
  '';
  nativeBuildInputs = kernel.moduleBuildDependencies;
  makeFlags = kernel.makeFlags ++ [
    "-C"
    "${kernel.dev}/lib/modules/${kernel.modDirVersion}/build"
    "M=$(sourceRoot)"
  ];
  buildFlags = [ "modules" ];
  installFlags = [ "INSTALL_MOD_PATH=${placeholder "out"}" ];
  installTargets = [ "modules_install" ]; 

  meta = with lib; {
    description = "Improved Linux driver for Acer RGB Keyboards ";
    homepage = "https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module";
    license = licenses.gpl3;
    maintainers = [ ];
    platforms = platforms.linux;
  };
}
```
Then, add this line after the initial { config, pkgs, ... }:
`let acermodule = config.boot.kernelPackages.callPackage ./acer-module.nix {}; in`

Afterwards, import the module package and the required modules on the same configuration.nix:
```
  boot.extraModulePackages = [ acermodule ];
  boot.kernelModules = [ "facer" "wmi" "sparse-keymap" "video" ];
```

Done, with that rebuild your system and reboot. The module should be built, and the ./facer.py should work to configure your laptop keyboard.

closed 2024-01-12 by AligatorRicardo

SpiderUnderUrBed, 2024-04-24:
Where would facer.py be built?

SpiderUnderUrBed, 2024-05-21:
Nvm solved, i just clone it and run facer.py