JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #137: Stopped working randomly

KaiserGabo, 2024-01-02:
# Model: 
Manufacturer: Acer
Product Name: Nitro AN515-58
Version: V2.03
Serial Number: NHQLZAA0013170FF373400
UUID: dd23cb4c-69e2-ed11-80ea-40c2ba150928
Wake-up Type: Power Switch
--
Base Board Information
Manufacturer: ADL
Product Name: Jimny_ADH
Version: V2.03
Serial Number: NBQLZ11001317312C33400
Asset Tag: Type2 - Board Asset Tag
Features:
`

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1

### RGB keyboard works?  No

It worked for a few days but then the RGB stopped. Tried reinstalling it but it doesn't mount as /dev/acer-gkbbl-0, instalation also doesn't show any errors

Argentum11, 2024-01-12:
@KaiserGabo Mine (product name: Predator PH315-53) stopped working this week, when I tired to control the RGB keyboard, i encountered the error 

```PermissionError: [Errno 13] Permission denied: '/dev/acer-gkbbl-0'```

Is this the same situation that you met?

RazoBlast, 2024-01-12:
Same issue here. Found a solution yet?

RazoBlast, 2024-01-12:
> @KaiserGabo Mine (product name: Predator PH315-53) stopped working this week, when I tired to control the RGB keyboard, i encountered the error
> 
> `PermissionError: [Errno 13] Permission denied: '/dev/acer-gkbbl-0'`
> 
> Is this the same situation that you met?

Yep, what is this

KaiserGabo, 2024-01-13:
> @KaiserGabo Mine (product name: Predator PH315-53) stopped working this week, when I tired to control the RGB keyboard, i encountered the error
> 
> `PermissionError: [Errno 13] Permission denied: '/dev/acer-gkbbl-0'`
> 
> Is this the same situation that you met?

The RGB GUI says `/dev/acer-gkbbl-0 `is not even mounted and needs to be installed. Idk what's happening.

JafarAkhondali, 2024-01-13:
By any chances, did upgrade the kernel recently? Maybe the new version is not compatible?

KaiserGabo, 2024-01-13:
> By any chances, did upgrade the kernel recently? Maybe the new version is not compatible? 

Just  downgraded to the oldest package I have in cache (haven't cleared the pkg cache at all since installing the OS), reinstalled from both source and the Arch Linux Repo, still nothing

KaiserGabo, 2024-01-16:
Update:
Just reinstalled arch linux all together. I'm on kernel `6.7.0-zen3-1-zen` and installed the package from pacman. It's working now. Idk what may have broke last time because I'm pretty sure I updated the kernel once and didn't broke the RGB, rolling back into the same kernel also didn't fix anything. For now I'm not updating the kernel to see if one of the packages I usually install does the magic trick.

RazoBlast, 2024-01-19:
Found the issue, recomplied it. I need to do this everytime though. Can I set up a pacman hook perhaps?

JafarAkhondali, 2024-01-19:
Related: https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/113

Argentum11, 2024-07-12:
@RazoBlast It seems that your module works after recompilation, is the recompilation done by uninstall and reinstall the whole module, or how should i recompile?