JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #152: Reporting patches in kernel 6.8 for cyclic modes in phn16-71

mitanshu7, 2024-05-30:
Please have a look and see if it will be helpful!
https://bbs.archlinux.org/viewtopic.php?id=291076

JafarAkhondali, 2024-06-12:
Thanks @mitanshu7 
Looks interesting, we really need to change the way this repository works to get updates from Linux kernel, until we can merge it completely.

mitanshu7, 2024-06-16:
Hi @JafarAkhondali 

I used this kernel on my machine (via Fedora 40) and the modes cycled fine, even the fan profiling work as the temps increase. What I miss is the RGB customization. Is there a way we can send a patch upstream? or maybe a simple python module should be enough?

Apologies for the vague suggestions, I don't know how much of this works, but want to help!

closed 2024-08-20 by mitanshu7