JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #153: Doesn't work in kernel 6.8

yusifvatan, 2024-06-04:
# Model: `phn16-71`

## Count of keyboard RGB zones: 4

### Count of Cpu Fans: 1
### Count of Gpu Fans: 1

### RGB keyboard works?  Yes
### Turbo button turn on fans?  Yes
### Turbo button turn on LED? No
### Turbo button activates overclock? I don't know

Hi. Change backlight and turbo mode works  in kernel 6.5. In kernel 6.8 update you can change mods (quite , balanced, performance , turbo). I updated kernel . But now I can't change keyboard backlight . Is there a way change backlight?

JafarAkhondali, 2024-06-04:
Hi,
How exactly does the backlight mode worked in kernel 6.5? how did you control rgb colors? was dynamic coloring working as well? I don't have an Acer anymore

yusifvatan, 2024-06-04:
Everything was working normally. When I wrote the mode/color at the end of the facer_rgb.py command, it worked

yusifvatan, 2024-06-04:
When I try install I got this error

$ sudo ./install.sh
make -C /lib/modules/6.8.1-060801-generic/build M=/home/yusifvatan/Desktop/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: *** /lib/modules/6.8.1-060801-generic/build: No such file or directory.  Stop.
make: *** [Makefile:14: default] Error 2
insmod: ERROR: could not load module src/facer.ko: No such file or directory
[   24.084844] rfkill: input handler enabled
[   25.492871] rfkill: input handler disabled
[  163.406287] acer_wmi: Acer Laptop ACPI-WMI Extras
[  163.406343] acer_wmi: Function bitmap for Communication Button: 0x801
[  163.419211] input: Acer WMI hotkeys as /devices/virtual/input/input23
[  746.139530] acer_wmi: Acer Laptop WMI Extras unloaded
[  870.808186] acer_wmi: Acer Laptop ACPI-WMI Extras
[  870.808216] acer_wmi: Function bitmap for Communication Button: 0x801
[  870.824249] input: Acer WMI hotkeys as /devices/virtual/input/input24
[  889.742268] acer_wmi: Acer Laptop WMI Extras unloaded

JafarAkhondali, 2024-06-04:
Ok there was a misunderstanding, I thought you were able to change colors without using this repository. Try install linux headers, uninstall the repository, reboot, install repo again

yusifvatan, 2024-06-05:
Thank for advice . But it does't work
I try everyhing.  I even downloaded the Ubuntu 24.04  but it didn't work

yusifvatan, 2024-07-14:
I downloaded pop os. I didn't expect at all but it worked. I have no idea what I did wrong before . Now I can change both  performance modes and the keyboard backlight . Thank for this repository

closed 2024-07-14 by yusifvatan