JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #154: Nitro AN16-41

Sasikuttan2163, 2024-06-08:
# Model: Nitro AN16-41

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1 

### RGB keyboard works?  Yes
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? No

The RGB module works really well. Something which needs to be documented is that entering the RGB mode higher than 5 activates the other effects exclusive to the new NitroSense such as Twinkling and Meteor. 

However turbo switch doesn't work. I tried going through `dmesg` and `journalctl -f` outputs but neither showed any of the warnings which should show up if the key is not detected. It would be great if you can tell me how to write a custom module for this.

giltwist, 2024-06-20:
Confirming that RGB lighting control works, including the [Plasma 6 Widget based on this](https://github.com/x211321/RGB-Config-Acer-gkbbl-0-Plasma-Widget-Plasma6).  However, the new [KDE Plasma 6.1 keyboard rgb sync feature](https://www.gamingonlinux.com/2024/04/kde-plasma-61-will-be-able-to-sync-keyboard-rgb-to-your-chosen-accent-colour/) does not appear to recognize the existence of the RGB functionality.

x211321, 2024-06-20:
> Confirming that RGB lighting control works, including the [Plasma 6 Widget based on this](https://github.com/x211321/RGB-Config-Acer-gkbbl-0-Plasma-Widget-Plasma6). However, the new [KDE Plasma 6.1 keyboard rgb sync feature](https://www.gamingonlinux.com/2024/04/kde-plasma-61-will-be-able-to-sync-keyboard-rgb-to-your-chosen-accent-colour/) does not appear to recognize the existence of the RGB functionality. Unsure if the issue is on this module's end (needing to report its existence in some way) or on kdeplasma-addon's end (needing to listen to this module).

This module implements it's own custom interface to control the RGB values, that's why my config app and plasma widget are specific for this module. 

I don't know how RGB interfaces usually look like but since this one appears to be non-standard there is probably no way to link it to general RGB control software like the one in plasma 6.1.

JafarAkhondali, 2024-06-20:
@giltwist this repository(mine), provides a custom interface character device and the python script interacts with it. The method I used is not standard, so KDE desktop also didn't detect this method.  
The standard method is to create an LED file, which then be placed in `/sys/class/leds/`. But the issue (at the time I wrote the module) was that there was no feature to implement all these features in a standard way. For example, you can't have custom effects, or control multiple zones at the same time, etc.
There are some new attempts:
https://github.com/torvalds/linux/blob/master/drivers/leds/rgb/leds-group-multicolor.c

You can see part of this long discussion here:
https://lore.kernel.org/all/cde716b4-450a-1639-5994-aabef643105e@metux.net/T/#m83f3c8bcf440f241a19cfad4c2b96ef40320d3be

and here:
https://www.spinics.net/lists/platform-driver-x86/msg25769.html

DetuxTR, 2024-07-20:
I`m writing a nitro fan speed controller module. You can see it from my profile.

Z-Diva, 2024-11-01:
hello i have the same model . Did you fix the turbo button?

Sasikuttan2163, 2024-11-03:
Nope

abduvaliy-engineer, 2026-10-04:
@Sasikuttan2163 @Z-Diva @DetuxTR since you have the Nitro AN16-41: does the built-in camera work for you on Linux?

The AN16-41 is an AMD (Ryzen) model, like mine. On my Nitro ANV16S-41 (Ryzen 7 260) the camera was missing completely (`lsusb` showed no camera, no `/dev/video0`). The cause was a GPIO pin that powers the camera, which Linux turns off at boot. Intel Nitros seem to work out of the box, so I'm trying to find out which AMD models are affected.

If your camera is missing too, this might fix it, and the `check.sh` script there shows whether it's the same problem:
https://github.com/abduvaliy-engineer/acer-nitro-anv16s-camera-fix

If your camera works fine, please say so too, that helps just as much.

Sasikuttan2163, 2026-10-05:
For me the camera worked out of the box didn't have to add anything myself