JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #158: Predator PHN18-71

blackPantherOS, 2024-06-18:
# Model: 

```
System Information
        Manufacturer: Acer
        Product Name: Predator PHN18-71
        Version: V1.11
        Serial Number: NHQNQEU00740406A******
        UUID: ddfbcd46-23b5-ee11-80f2-40c2b******
        Wake-up Type: Power Switch

Base Board Information
        Manufacturer: RPL
        Product Name: IPACE_RTX
        Version: V1.11
        Serial Number: NBQQU110024044C6******
        Asset Tag: Type2 - Board Asset Tag
        Features:
```

## Count of keyboard RGB zones: 
4

### Count of Cpu Fans:  
1
### Count of Gpu Fans: 
1 
### RGB keyboard works?  
Yes
### Turbo button turn on fans?  
No
### Turbo button turn on LED? 
No
### Turbo button activates overclock? 
No
[dsdt.tar.gz](https://github.com/user-attachments/files/15889619/dsdt.tar.gz)

mitanshu7, 2024-06-23:
Hello, you may want to look at this [commit](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/pull/149/commits/3cd640424847694bdec9ebbf942451540a0f94db). I basically added my model specs there and it worked. You can try replacing the 16 with 18 and it should work.

blackPantherOS, 2024-06-25:
> Hello, you may want to look at this [commit](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/pull/149/commits/3cd640424847694bdec9ebbf942451540a0f94db). I basically added my model specs there and it worked. You can try replacing the 16 with 18 and it should work.

Hi, okay thanks, this week I will try to add these modifications to facer source code as well.
I built a custom kernel from 6.9.5 with many acer_wmi modifications, and many things are working now from gui.

I forked [this](https://github.com/AstromanGaming/Linux-PredatorSense-Fork.git) repo and I working on new sense interface gui.

[image]

JafarAkhondali, 2024-06-25:
@blackPantherOS This looks pretty nice 🔥 
Why there are so many forks of this project? it would be nice if we could keep only one fork so that people can contribute to one project.

Megamichi123, 2024-08-27:
can you please merge this? I have a new phn18 and would like to install the module, how do I do that?

blackPantherOS, 2024-08-27:
Of course guys, but I still did not have free time to finish the application. If someone has time to work on it, I'll be happy to pull it to our github

Megamichi123, 2024-08-27:
I have now installed the drivers and they work perfectly :)

mmsaeed509, 2024-08-30:
Hi, guys!
I'm working on a GUI App that replicates the look and feel of the PredatorSense app. I uploaded the src code today and You can find out here: [PredatorSense-Linux](https://github.com/mmsaeed509/PredatorSense-Linux).

[image]

This project will take some time to complete due to my current schedule, as I have no time. However, once it's finished, I plan to merge all existing functionalities from the current PredatorSense application by @blackPantherOS into this project. The goal is to establish this app as the official version of Linux, encouraging community contributions and improvements on this app.

---

**Note:** There is also a special edition of Exodia OS tailored specifically for Acer Predator and Nitro laptops, called the Acer-Predator Edition. This edition includes this kernel module pre-installed. you can find out here [Exodia OS Acer-Predator Edition](https://exodia-os.github.io/download.html).

[image]