JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #166: Kernel 6.10.2 is breaking the installation (again)

xAlpharax, 2024-07-29:
# Model: AN515-45

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1 

### RGB keyboard works?  No
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? No

I get this output now with the latest stable kernel (6.10.2):

alphara@deltarion ~/dev/acer-rgb-turbo> sudo ./install.sh
make -C /lib/modules/6.10.2_1/build M=/home/alphara/dev/acer-rgb-turbo modules
make[1]: Entering directory '/usr/src/kernel-headers-6.10.2_1'
make[1]: Leaving directory '/usr/src/kernel-headers-6.10.2_1'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
    sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
rmmod: ERROR: Module acer_wmi is not currently loaded
insmod: ERROR: could not insert module src/facer.ko: Invalid module format
[   24.374348] Bluetooth: RFCOMM TTY layer initialized
[   24.374367] Bluetooth: RFCOMM socket layer initialized
[   24.374373] Bluetooth: RFCOMM ver 1.11
[   72.540964] facer: version magic '6.10.1_1 SMP preempt mod_unload ' should be '6.10.2_1 SMP preempt mod_unload '
[   90.619968] acer_wmi: Acer Laptop ACPI-WMI Extras
[   90.619999] acer_wmi: Function bitmap for Communication Button: 0x801
[   90.625471] input: Acer WMI hotkeys as /devices/virtual/input/input19
[   93.182655] acer_wmi: Acer Laptop WMI Extras unloaded
[   93.204850] facer: version magic '6.10.1_1 SMP preempt mod_unload ' should be '6.10.2_1 SMP preempt mod_unload '
[  128.326417] facer: version magic '6.10.1_1 SMP preempt mod_unload ' should be '6.10.2_1 SMP preempt mod_unload '
[*] Done
alphara@deltarion ~/dev/acer-rgb-turbo>

It seems that the kernel modules have been screwed up again.

JafarAkhondali, 2024-07-30:
I don't know 🫤  maybe you disabled the module signing? Anyways if it's solved then maybe it's better to keep it closed and open it again if it happened again.

xAlpharax, 2024-07-31:
> I don't know 🫤  maybe you disabled the module signing? Anyways if it's solved then maybe it's better to keep it closed and open it again if it happened again.

will do 🫡

closed 2024-07-31 by xAlpharax