JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #177: Install fails on Nobara Linux 6.11.0-200.fsync.fc40.x86_64

LuisMayo, 2024-09-21:
# Model:
Nitro ANV16-41

## Count of keyboard RGB zones
Unsure since I never used Windows on this, however the `/sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/` directory exists

### Count of Cpu Fans:
Unknown
### Count of Gpu Fans
Unkown

### RGB keyboard works?
Install fails
### Turbo button turn on fans?
Install fails
### Turbo button turn on LED?
Install fails
### Turbo button activates overclock?
Install Fails

The `install.sh` fails, I copy the stacktrace here:
```
/home/luis/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:3102:27: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Wincompatible-pointer-types]
 3102 |                 .remove = acer_platform_remove,
      |                           ^~~~~~~~~~~~~~~~~~~~
```

This is probably due to the kernel version being a branched version from `6.11`  with fsync support, I haven't tested with official 6.11 but maybe it doesn't work either. If the problem is the branched kernel this can be closed, but I'm opening in case the problem is the `6.11`kernel itself, or something wrong on my end.

giltwist, 2024-09-23:
I don't think it's just your specific branched kernel.  I'm getting the same error on 6.11.0-5-cachyos.

LuisMayo, 2024-09-23:
I did test with 6.10 with fsync branch and the install worked (nothing else did though). So we now know 2 things

Kernel 6.11 doesn't work-
Nitro ANV16-41 install but doesn't work

Zeronymus, 2024-09-25:
Mine had the same issue on the latest nobara kernel

groopcd, 2024-10-01:
same issue here, kernel 6.11 opensuse tumbleweed

ct 01 10:00:39 kerberos bash[1282]: make -C /lib/modules/6.11.0-1-default/build M=/opt/turbo-fan modules
Oct 01 10:00:39 kerberos bash[1305]: make[1]: Entering directory '/usr/src/linux-6.11.0-1-obj/x86_64/default'
Oct 01 10:00:40 kerberos bash[1657]:   CC [M]  /opt/turbo-fan/src/facer.o
Oct 01 10:00:40 kerberos bash[1659]: /opt/turbo-fan/src/facer.c:1946:6: warning: no previous prototype for ‘WMID_gaming_set_fan_mode’ [-Wmissing-prototypes]
Oct 01 10:00:40 kerberos bash[1659]:  1946 | void WMID_gaming_set_fan_mode(u8 fan_mode)
Oct 01 10:00:40 kerberos bash[1659]:       |      ^~~~~~~~~~~~~~~~~~~~~~~~
Oct 01 10:00:40 kerberos bash[1659]: /opt/turbo-fan/src/facer.c:3102:27: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Wincompatible-pointer-types]
Oct 01 10:00:40 kerberos bash[1659]:  3102 |                 .remove = acer_platform_remove,
Oct 01 10:00:40 kerberos bash[1659]:       |                           ^~~~~~~~~~~~~~~~~~~~
Oct 01 10:00:40 kerberos bash[1659]: /opt/turbo-fan/src/facer.c:3102:27: note: (near initialization for ‘acer_platform_driver.<anonymous>.remove’)
Oct 01 10:00:40 kerberos bash[1659]: /opt/turbo-fan/src/facer.c:1921:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
Oct 01 10:00:40 kerberos bash[1659]:  1921 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
Oct 01 10:00:40 kerberos bash[1659]:       |                    ^~~~~~~~~~~~~~~~~~~
Oct 01 10:00:40 kerberos bash[1654]: make[3]: *** [/usr/src/linux-6.11.0-1/scripts/Makefile.build:245: /opt/turbo-fan/src/facer.o] Error 1
Oct 01 10:00:40 kerberos bash[1308]: make[2]: *** [/usr/src/linux-6.11.0-1/Makefile:1949: /opt/turbo-fan] Error 2
Oct 01 10:00:40 kerberos bash[1305]: make[1]: *** [../../../linux-6.11.0-1/Makefile:224: __sub-make] Error 2
Oct 01 10:00:40 kerberos bash[1305]: make[1]: Leaving directory '/usr/src/linux-6.11.0-1-obj/x86_64/default'
Oct 01 10:00:40 kerberos bash[1282]: make: *** [Makefile:14: default] Error 2
Oct 01 10:00:41 kerberos bash[1748]: insmod: ERROR: could not load module src/facer.ko: No such file or directory