JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #181: Kernel 6.11 magic changes error the install

xAlpharax, 2024-10-01:
# Model: `sudo dmidecode | grep "Product Name" -B 2 -A 4`
Product Name: Nitro AN515-45

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1 

### RGB keyboard works?  No
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? No

alphara@deltarion ~/dev/acer-rgb-turbo> sudo ./install.sh
make -C /lib/modules/6.11.0_1/build M=/home/alphara/dev/acer-rgb-turbo modules
make[1]: Entering directory '/usr/src/kernel-headers-6.11.0_1'
make[1]: Leaving directory '/usr/src/kernel-headers-6.11.0_1'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
    sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
insmod: ERROR: could not insert module src/facer.ko: Invalid module format
[   95.139649] facer: version magic '6.10.11_1 SMP preempt mod_unload ' should be '6.11.0_1 SMP preempt mod_unload '
[  129.791184] r8169 0000:04:00.0 enp4s0: Link is Down
[  133.417209] r8169 0000:04:00.0 enp4s0: Link is Up - 1Gbps/Full - flow control rx/tx
[  133.767264] r8169 0000:04:00.0 enp4s0: Link is Down
[  141.390027] r8169 0000:04:00.0 enp4s0: Link is Up - 1Gbps/Full - flow control rx/tx
[  141.771044] acer_wmi: Acer Laptop ACPI-WMI Extras
[  141.771076] acer_wmi: Function bitmap for Communication Button: 0x801
[  141.777068] input: Acer WMI hotkeys as /devices/virtual/input/input37
[  145.207396] acer_wmi: Acer Laptop WMI Extras unloaded
[  145.235846] facer: version magic '6.10.11_1 SMP preempt mod_unload ' should be '6.11.0_1 SMP preempt mod_unload '
[*] Done

LarryTheMagicDragon, 2024-10-02:
I think it's a minor change in the kernel code making the module fail to compile. I was able to get it working on my system and I've open a pull request with the fix

xAlpharax, 2024-10-05:
Good to hear!, hopefully it s merged soon.

LarryTheMagicDragon, 2024-10-07:
It has been, this ought to be marked as closed.