JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #197: Cannot compile in 6.11 kernel

davidwkr, 2024-10-19:
make -C /lib/modules/6.11.4-4-cachyos/build M=/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.11.4-4-cachyos/build'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: clang version 18.1.8
  You are using:           gcc (GCC) 14.2.1 20240910
  CC [M]  /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
make[3]: *** [scripts/Makefile.build:244: /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Error 1
make[2]: *** [/usr/lib/modules/6.11.4-4-cachyos/build/Makefile:1934: /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.11.4-4-cachyos/build'
make: *** [Makefile:14: default] Error 2

LarryTheMagicDragon, 2024-10-19:
The gcc errors indicate you are passing the flags -mretpoline-external-thunk and -fsplit-lto-unit to gcc somehow. I cannot recreate on my system, but I believe your version of make is configured to pass those options to your version of gcc. Try adding CC=clang to the top of your makefile and see what happens

LarryTheMagicDragon, 2024-10-19:
Actually I just remembered you don't need to edit the makefile, you can just try running 'make CC=clang' and it will have the same effect

davidwkr, 2024-10-19:
> Actually I just remembered you don't need to edit the makefile, you can just try running 'make CC=clang' and it will have the same effect
Ty. Tried but didn't work

```
make -C /lib/modules/6.11.4-4-cachyos/build M=/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.11.4-4-cachyos/build'
  CC [M]  /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1977:20: warning: unused function 'WMID_gaming_get_u64' [-Wunused-function]
 1977 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
1 warning generated.
ld: unrecognised emulation mode: llvm
Supported emulations: elf_x86_64 elf32_x86_64 elf_i386 elf_iamcu i386pep i386pe elf64bpf
make[3]: *** [scripts/Makefile.build:244: /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Error 1
make[3]: *** Deleting file '/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o'
make[2]: *** [/usr/lib/modules/6.11.4-4-cachyos/build/Makefile:1934: /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.11.4-4-cachyos/build'
make: *** [Makefile:14: default] Error 2

```

LarryTheMagicDragon, 2024-10-19:
can you post 'make CC=clang V=1'? That will give more detailed information.

davidwkr, 2024-10-19:
> can you post 'make CC=clang V=1'? That will give more detailed information.

Sure

```
make -C /lib/modules/6.11.4-4-cachyos/build M=/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.11.4-4-cachyos/build'
make --no-print-directory -C /usr/lib/modules/6.11.4-4-cachyos/build \
-f /usr/lib/modules/6.11.4-4-cachyos/build/Makefile modules
make -f ./scripts/Makefile.build obj=/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module need-builtin=1 need-modorder=1 
# CC [M]  /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
  clang -Wp,-MMD,/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/.facer.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ --target=x86_64-linux-gnu -fintegrated-as -Werror=ignored-optimization-argument -Werror=option-ignored -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -mno-avx2 -mno-avx512f -fno-tree-vectorize -fcf-protection=branch -fno-jump-tables -m64 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mstack-alignment=8 -mskip-rax-setup -march=x86-64-v3 -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mretpoline-external-thunk -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O3 -fstack-protector-strong -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mfentry -DCC_USING_NOP_MCOUNT -DCC_USING_FENTRY -fno-lto -flto=thin -fsplit-lto-unit -fvisibility=hidden -falign-functions=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wframe-larger-than=2048 -Wno-gnu -Wvla -Wno-pointer-sign -Wcast-function-type -Wimplicit-fallthrough -Werror=date-time -Wenum-conversion -Wextra -Wunused -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-format-overflow -Wno-format-overflow-non-kprintf -Wno-format-truncation-non-kprintf -Wno-override-init -Wno-pointer-to-enum-cast -Wno-tautological-constant-out-of-range-compare -Wno-unaligned-access -Wno-enum-compare-conditional -Wno-enum-enum-conversion -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-sign-compare -Wno-unused-parameter -g -gdwarf-5  -DMODULE  -DKBUILD_BASENAME='"facer"' -DKBUILD_MODNAME='"facer"' -D__KBUILD_MODNAME=kmod_facer -c -o /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c  ; ld -m elf_x86_64 -mllvm -import-instr-limit=5 -z noexecstack   -r -o /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/.tmp_facer.o /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o; mv /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/.tmp_facer.o /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o  ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --hacks=skylake --ibt --mcount --mnop --orc --retpoline --rethunk --sls --static-call --uaccess --prefix=16  --link  --module /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1977:20: warning: unused function 'WMID_gaming_get_u64' [-Wunused-function]
 1977 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
1 warning generated.
ld: unrecognised emulation mode: llvm
Supported emulations: elf_x86_64 elf32_x86_64 elf_i386 elf_iamcu i386pep i386pe elf64bpf
make[3]: *** [scripts/Makefile.build:244: /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Error 1
make[3]: *** Deleting file '/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o'
make[2]: *** [/usr/lib/modules/6.11.4-4-cachyos/build/Makefile:1934: /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.11.4-4-cachyos/build'
make: *** [Makefile:14: default] Error 2

```

LarryTheMagicDragon, 2024-10-19:
okay, I think this is being caused by the gcc linker being used instead of the clang linker, try 'make CC=clang LD=ld.lld V=1' I don't know if you have the lld package for your os, but it will fail without it.

davidwkr, 2024-10-19:
> okay, I think this is being caused by the gcc linker being used instead of the clang linker, try 'make CC=clang LD=ld.lld V=1' I don't know if you have the lld package for your os, but it will fail without it.
I do have lld. You nailed it with those options, it worked! :) I was missing it so much, glad to have it back. Thank you

LarryTheMagicDragon, 2024-10-19:
I'm glad I could help.

closed 2024-10-19 by davidwkr