JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #198: Breaks Power Profile in KDE

SaxatB, 2024-10-22:
Model : PHN16-71

Before installing this the power profile was working fine but now it gives me a error.

KrisDevelopment, 2024-10-22:
Can you share what the error is and a little more details about the system?

SaxatB, 2024-10-23:
> Can you share what the error is and a little more details about the system?

Keyboard RGB Zones - 4
CPU Fans - 1
GPU Fans - 1
RGB Keyboard Works? Yes.

Previously, using KDE's Power Profile option used to change power profile from eco, balanced and performance. After installing this module, while the RGB is more customizable I'm no longer able to change power profile. If I try chaning via powerprofilesctl, it gives me a error that reads "Failed to communicate with power-profiles-daemon: g-io-error-quark: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code1: Failed to activate platform driver 'platform_profile': Failed to write to acpi_platform_profile: Could not open '/sys/firmware/acpi/platform_profile' for writing (1)".

The fans also seem to be at full speed when opening the laptop but it gets fixed once I log in.

SaxatB, 2024-10-24:
> > > Can you share what the error is and a little more details about the system?
> > 
> > 
> > Keyboard RGB Zones - 4 CPU Fans - 1 GPU Fans - 1 RGB Keyboard Works? Yes.
> > Previously, using KDE's Power Profile option used to change power profile from eco, balanced and performance. After installing this module, while the RGB is more customizable I'm no longer able to change power profile. If I try chaning via powerprofilesctl, it gives me a error that reads "Failed to communicate with power-profiles-daemon: g-io-error-quark: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code1: Failed to activate platform driver 'platform_profile': Failed to write to acpi_platform_profile: Could not open '/sys/firmware/acpi/platform_profile' for writing (1)".
> > The fans also seem to be at full speed when opening the laptop but it gets fixed once I log in.
> 
> I'm also on KDE, but can't get it to break for me (maybe I don't run the same power profile app as you), however I'm sure platform_profile requires exporting the proper kernel module. Can you run this: `grep platform_profile_register /proc/kallsyms` Output shouldn't be empty, it should give you something like:
> 
> ```
> 0000000000000000 r __crc_platform_profile_register      [platform_profile]
> 0000000000000000 r __kstrtab_platform_profile_register  [platform_profile]
> 0000000000000000 r __kstrtabns_platform_profile_register        [platform_profile]
> 0000000000000000 r __ksymtab_platform_profile_register  [platform_profile]
> 0000000000000000 t platform_profile_register    [platform_profile]
> ```

I got this as the output:
"0000000000000000 r __kstrtab_platform_profile_register  [platform_profile]
0000000000000000 r __kstrtabns_platform_profile_register        [platform_profile]
0000000000000000 r __ksymtab_platform_profile_register  [platform_profile]
0000000000000000 t __pfx_platform_profile_register      [platform_profile]
0000000000000000 t platform_profile_register    [platform_profile]"