JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #199: Acer Predator Neo 16 - not working

abruenin, 2024-10-29:
# Model:

```
System Information
        Manufacturer: Acer
        Product Name: Predator PH16-71
        Version: V1.16
        Serial Number: NHQJQEG0033100B5A77600
        UUID: a5b635d8-d893-a047-9f22-c01850d93328
        Wake-up Type: Power Switch
--
Base Board Information
        Manufacturer: RPL
        Product Name: Discovery_RTX
        Version: V1.16
        Serial Number: NBQJQ110013100B5A77600
        Asset Tag: Type2 - Board Asset Tag
        Features:
```
## Count of keyboard RGB zones: 4

### RGB keyboard works?  No
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? No

I saw another post here, that it is working on the same model. But it is not working at all here (Manjaro). The Neo 16 has these randomly blinking keys at startup and these remain blinking. But I have the same model

```
sudo dmidecode -s system-product-name
Predator PH16-71
```

and I have checked, that the directory exist

```
file /sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/
/sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/: directory
```

The acer_wmi module is unloaded and I have the facer module loaded:

```lsmod | grep acer

facer                  53248  0
platform_profile       12288  1 facer
i8042                  57344  1 facer
sparse_keymap          12288  1 facer
rfkill                 40960  10 iwlmvm,bluetooth,facer,cfg80211
video                  81920  5 nvidia_wmi_ec_backlight,facer,xe,i915,nvidia_modeset
wmi                    32768  4 video,nvidia_wmi_ec_backlight,wmi_bmof,facer
```

The character device is there:

```
ls /dev/acer*
/dev/acer-gkbbl-0  /dev/acer-gkbbl-static-0
```

But no matter, what I configure with `keyboard.py` or `facer_rgb.py`, I only have the randomly blinking keys.

Sudococommunity, 2024-11-20:
how you control the temp of lapi

Sudococommunity, 2024-11-20:
i am searching for fan control do you got any

veryfunnyaz, 2024-12-30:
Were you somehow able to fix this?

abruenin, 2024-12-30:
> Were you somehow able to fix this?

Unfortunately not, still need a solution