JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #200: RGB Keyboard not working on lts kernel arch linux

DJPretzel-bit64, 2024-11-04:
# Model: `Nitro AN515-45`

## Count of keyboard RGB zones: 4

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1 

### RGB keyboard works?  Yes
### Turbo button turn on fans?  Not Tested
### Turbo button turn on LED? Not Tested
### Turbo button activates overclock? Not Tested

I'm using x211321's Plasma 6 Applet to interface with the module, but it only seems to work on the regular linux kernel. When I boot with the lts kernel it gives me an error saying that the keyboard device doesn't exist. I have to use the lts kernel because there is currently a problem with the main kernel not waking from suspend  when using Nvidia drivers. I installed this via the ExpediaOS Repo. Is there a way to install it for the linux-lts kernel manually?

DJPretzel-bit64, 2024-11-04:
I tried going through the manual install (one-time), and when I run `sudo ./install.sh` it gives the error "insmod: ERROR: could not insert module src/facer.ko: Unknown symbol in module". I tried it on both the lts kernel and the regular kernel, and it works on the regular kernel.

marvinhrivera, 2024-11-04:
Same here, I'm using a PH315-55 though, but same effect that yours, if I use the current 6.11.6 kernel, it works, however if I use the LTS Kernel 6.6.59 it stops working. 

For a temporary fix I reverted back to 1186522 commit for it to work again.

JafarAkhondali, 2024-11-04:
@LarryTheMagicDragon Can you check this please?

LarryTheMagicDragon, 2024-11-05:
Can confirm this issue with kernel 6.6.59-1-lts. when I ran install.sh this was the output
```
make -C /lib/modules/6.6.59-1-lts/build M=/storage/ltmd/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.6.59-1-lts/build'
  CC [M]  /storage/ltmd/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/storage/ltmd/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1977:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
 1977 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
  MODPOST /storage/ltmd/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/Module.symvers
  CC [M]  /storage/ltmd/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.mod.o
  LD [M]  /storage/ltmd/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
  BTF [M] /storage/ltmd/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
make[1]: Leaving directory '/usr/lib/modules/6.6.59-1-lts/build'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
        sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
rmmod: ERROR: Module acer_wmi is not currently loaded
insmod: ERROR: could not insert module src/facer.ko: Unknown symbol in module
[ 1097.801331] facer: Unknown symbol platform_profile_remove (err -2)
[ 1415.694047] facer: Unknown symbol platform_profile_register (err -2)
[ 1415.694120] facer: Unknown symbol platform_profile_notify (err -2)
[ 1415.694216] facer: Unknown symbol platform_profile_remove (err -2)
[ 1422.018858] facer: Unknown symbol platform_profile_register (err -2)
[ 1422.018938] facer: Unknown symbol platform_profile_notify (err -2)
[ 1422.019043] facer: Unknown symbol platform_profile_remove (err -2)
[ 1454.886946] facer: Unknown symbol platform_profile_register (err -2)
[ 1454.887024] facer: Unknown symbol platform_profile_notify (err -2)
[ 1454.887128] facer: Unknown symbol platform_profile_remove (err -2)
[*] Done
```

4d7d033 is the latest commit that works. these kinds of errors start with 8bb294a
I've looked at the code before and after, and as far as I can tell all the (err -2) are from those functions not being in the LTS headers. I was able to temporarily fix this by adding these starting at line 1163
```
int platform_profile_remove()
{
	return 0;
}

int platform_profile_register(struct platform_profile_handler* platform_profile_handler)
{
	return 0;
}

void platform_profile_notify()
{
}
```
It still puts out a bunch of errors, but it still seems to work with these modifications
I'm not sure if there is a way to do conditional function declaration at runtime, The only other fix I can think of long term is some kind of conditional compilation based on whether or not you're using a LTS kernel.

LarryTheMagicDragon, 2024-11-05:
#202 is a patch for this problem

DJPretzel-bit64, 2024-11-05:
> #202 is a patch for this problem

yes, this also works for me, thank you!

marvinhrivera, 2024-11-05:
Indeed applying #202 fixes the issue for me as well. Thanks!