JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #204: Not compiling in 6.12

davidwkr, 2024-11-20:
Trying to compile in 6.12

`make -C /lib/modules/6.12.0-2-cachyos/build M=/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.12.0-2-cachyos/build'
make --no-print-directory -C /usr/lib/modules/6.12.0-2-cachyos/build \
-f /usr/lib/modules/6.12.0-2-cachyos/build/Makefile modules
make -f ./scripts/Makefile.build obj=/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module need-builtin=1 need-modorder=1 
# CC [M]  /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
  clang -Wp,-MMD,/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/.facer.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ --target=x86_64-linux-gnu -fintegrated-as -Werror=ignored-optimization-argument -Werror=option-ignored -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mstack-alignment=8 -mskip-rax-setup -march=x86-64-v3 -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mretpoline-external-thunk -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O3 -fstack-protector-strong -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mfentry -DCC_USING_NOP_MCOUNT -DCC_USING_FENTRY -fno-lto -flto=thin -fsplit-lto-unit -fvisibility=hidden -falign-functions=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wframe-larger-than=2048 -Wno-gnu -Wvla -Wno-pointer-sign -Wcast-function-type -Wimplicit-fallthrough -Werror=date-time -Wenum-conversion -Wextra -Wunused -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-format-overflow -Wno-format-overflow-non-kprintf -Wno-format-truncation-non-kprintf -Wno-override-init -Wno-pointer-to-enum-cast -Wno-tautological-constant-out-of-range-compare -Wno-unaligned-access -Wno-enum-compare-conditional -Wno-enum-enum-conversion -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-sign-compare -Wno-unused-parameter -g -gdwarf-5  -DMODULE  -DKBUILD_BASENAME='"facer"' -DKBUILD_MODNAME='"facer"' -D__KBUILD_MODNAME=kmod_facer -c -o /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c  ; ld.lld -m elf_x86_64 -mllvm -import-instr-limit=5 -z noexecstack   -r -o /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/.tmp_facer.o /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o; mv /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/.tmp_facer.o /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o  ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --hacks=skylake --ibt --mcount --mnop --orc --retpoline --rethunk --sls --static-call --uaccess --prefix=16  --link  --module /home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:2966:11: error: call to undeclared function 'wmi_get_event_data'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
 2966 |         status = wmi_get_event_data(value, &response);
      |                  ^
/home/black/apps/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:3219:7: error: incompatible function pointer types passing 'void (u32, void *)' (aka 'void (unsigned int, void *)') to parameter of type 'wmi_notify_handler' (aka 'void (*)(union acpi_object *, void *)') [-Wincompatible-function-pointer-types]
 3219 |                                                 acer_wmi_notify, NULL);
      |                                                 ^~~~~~~~~~~~~~~
./include/linux/acpi.h:404:25: note: passing argument to parameter 'handler' here
  404 |                                         wmi_notify_handler handler, void *data);
      |                                                            ^
2 errors generated.
`

LarryTheMagicDragon, 2024-11-20:
I just noticed that you've had this issue before, can you show the output of "cat /proc/version" ? I may be able to modify the install script to automatically resolve this issue

davidwkr, 2024-11-20:
❯ cat /proc/version
Linux version 6.12.0-2-cachyos (linux-cachyos@cachyos) (clang version 18.1.8, LLD 18.1.8) #1 SMP PREEMPT_DYNAMIC Mon, 18 Nov 2024 18:02:00 +0000

LarryTheMagicDragon, 2024-11-23:
I've made modifications. I'd like you to test #206  and make sure it works. I don't see any reason why it shouldn't, but I'd still like to have confirmation.

jyash8, 2024-11-24:
I am on kernel 6.12.1, and it`s not compiling and i am getting similar errors but i am using gcc compiler
i am on arch using this kernel - 6.12.1-arch1-1. These are the error logs:
```
make -C /lib/modules/6.12.1-arch1-1/build M=/home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.12.1-arch1-1/build'
  CC [M]  /home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘acer_wmi_notify’:
/home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:2966:18: error: implicit declaration of function ‘wmi_get_event_data’ [-Wimplicit-function-declaration]
 2966 |         status = wmi_get_event_data(value, &response);
      |                  ^~~~~~~~~~~~~~~~~~
/home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘acer_wmi_input_setup’:
/home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:3219:49: error: passing argument 2 of ‘wmi_install_notify_handler’ from incompatible pointer type [-Wincompatible-pointer-types]
 3219 |                                                 acer_wmi_notify, NULL);
      |                                                 ^~~~~~~~~~~~~~~
      |                                                 |
      |                                                 void (*)(u32,  void *) {aka void (*)(unsigned int,  void *)}
In file included from ./include/linux/i2c.h:13,
                 from ./include/uapi/linux/fb.h:6,
                 from ./include/linux/fb.h:5,
                 from /home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:22:
./include/linux/acpi.h:404:60: note: expected ‘wmi_notify_handler’ {aka ‘void (*)(union acpi_object *, void *)’} but argument is of type ‘void (*)(u32,  void *)’ {aka ‘void (*)(unsigned int,  void *)’}
  404 |                                         wmi_notify_handler handler, void *data);
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~
/home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: At top level:
/home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1977:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
 1977 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:229: /home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Error 1
make[2]: *** [/usr/lib/modules/6.12.1-arch1-1/build/Makefile:1936: /home/jyash8/acer-predator-turbo-and-rgb-keyboard-linux-module] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.12.1-arch1-1/build'
make: *** [Makefile:14: default] Error 2
```

Output of `cat /proc/version`:
```
Linux version 6.12.1-arch1-1 (linux@archlinux) (gcc (GCC) 14.2.1 20240910, GNU ld (GNU Binutils) 2.43.0) #1 SMP PREEMPT_DYNAMIC Fri, 22 Nov 2024 16:04:27 +0000
```

LarryTheMagicDragon, 2024-11-24:
#184 should work now.

davidwkr, 2024-11-24:
> I've made modifications. I'd like you to test #206 and make sure it works. I don't see any reason why it shouldn't, but I'd still like to have confirmation.
It didn't work with just that script change as I was already compiling with clang.

> #184 should work now.
Yes. It works for me in 6.12 with the changes in that PR. Thank you!!!

Valerio1474, 2024-11-24:
> #184 should work now.
It works on 6.12, thanks for your patch.

LarryTheMagicDragon, 2024-11-24:
@davidwkr @Valerio1474 I'm glad I could help

yhanruzai, 2024-12-01:
This error is not fixed with #184, at least after updating my NixOS from 24.05 to 24.11 which considering the other posts seem to be caused by the kernel update. It seems to print a similar error to what @jyash8 already said in this conversation