JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #205: New Model Predator PT14-51

J602, 2024-11-22:
# Model: System Information
	Manufacturer: Acer
	Product Name: Predator PT14-51
	Version: V1.05
	Serial Number: NHQLQEU00332700B887600
	UUID: 839bab54-41cc-7e41-88e0-78ddcb61d597
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: RPL
	Product Name: EV6_RTH
	Version: V1.05
	Serial Number: NBQLQ1100232700B887600
	Asset Tag: Type2 - Board Asset Tag
	Features:
## Count of keyboard RGB zones: 2

### Count of Cpu Fans:  1
### Count of Gpu Fans: 1  

### RGB keyboard works?  No
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? No
[dsdt.zip](https://github.com/user-attachments/files/17871873/dsdt.zip)

```
user@laptop~/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo dmidecode -s system-product-name
Predator PT14-51
user@laptop~/acer-predator-turbo-and-rgb-keyboard-linux-module$ sudo ./install.sh
make -C /lib/modules/6.1.0-27-amd64/build M=/home/user/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-27-amd64'
  CC [M]  /home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:3473:17: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 3473 |         .info = acer_wmi_hwmon_info,
      |                 ^~~~~~~~~~~~~~~~~~~
/home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1977:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
 1977 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
  MODPOST /home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/Module.symvers
  LD [M]  /home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
  BTF [M] /home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
Skipping BTF generation for /home/user/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-27-amd64'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
insmod: ERROR: could not insert module src/facer.ko: Unknown symbol in module
[ 4611.768091] facer: Unknown symbol platform_profile_register (err -2)
[ 4611.768122] facer: Unknown symbol platform_profile_notify (err -2)
[ 4611.768169] facer: Unknown symbol platform_profile_remove (err -2)
[ 4805.477613] acer_wmi: Acer Laptop ACPI-WMI Extras
[ 4805.477646] acer_wmi: Function bitmap for Communication Button: 0x801
[ 4805.488810] input: Acer WMI hotkeys as /devices/virtual/input/input42
[ 4826.905994] acer_wmi: Acer Laptop WMI Extras unloaded
[ 4826.960169] facer: Unknown symbol platform_profile_register (err -2)
[ 4826.960232] facer: Unknown symbol platform_profile_notify (err -2)
[ 4826.960330] facer: Unknown symbol platform_profile_remove (err -2)
[*] Done
```

LarryTheMagicDragon, 2024-11-23:
Are you using a LTS kernel? if you are try #202

J602, 2024-12-19:
sudo ./install.sh 
make -C /lib/modules/6.11.10-amd64/build M=/home/user/somedata/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-6.11.10-amd64'
make[1]: Leaving directory '/usr/src/linux-headers-6.11.10-amd64'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
rmmod: ERROR: Module acer_wmi is not currently loaded
[ 1789.911624] facer: Function bitmap for Communication Button: 0x801
[ 1789.927746] input: Acer WMI hotkeys as /devices/virtual/input/input10
[ 1795.688964] facer: Acer Laptop WMI Extras unloaded
[ 1795.775863] facer: Acer Laptop ACPI-WMI Extras
[ 1795.775919] facer: Function bitmap for Communication Button: 0x801
[ 1795.818647] input: Acer WMI hotkeys as /devices/virtual/input/input11
[ 1799.617422] facer: Acer Laptop WMI Extras unloaded
[ 1799.719378] facer: Acer Laptop ACPI-WMI Extras
[ 1799.719437] facer: Function bitmap for Communication Button: 0x801
[ 1799.738385] input: Acer WMI hotkeys as /devices/virtual/input/input12
[*] Done

LarryTheMagicDragon, 2024-12-19:
@J602 This output looks like it was successfully loaded. Does it work, or are you still having issues?