JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #207: Cannot install after Update

Reynardd, 2024-11-25:
# Model:System Information
	Manufacturer: Acer
	Product Name: Nitro AN515-45
	Version: V1.11
	Serial Number: NHQBCEM005202106163400
	UUID: a89972f9-1071-ec11-80e3-088fc33bcacc
	Wake-up Type: Power Switch
```
Base Board Information
	Manufacturer: CZ
	Product Name: Scala_CAS
	Version: V1.11
	Serial Number: NBQBC110022028C59D3400
	Asset Tag: Type2 - Board Asset Tag
	Features:
```
## Count of keyboard RGB zones: 1
### Count of Cpu Fans:  1
### Count of Gpu Fans: 1 
### RGB keyboard works?  No

make -C /lib/modules/6.12.1-arch1-1/build M=/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.12.1-arch1-1/build'
  CC [M]  /home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘acer_wmi_notify’:
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:2966:18: error: implicit declaration of function ‘wmi_get_event_data’ [-Wimplicit-function-declaration]
 2966 |         status = wmi_get_event_data(value, &response);
      |                  ^~~~~~~~~~~~~~~~~~
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘acer_wmi_input_setup’:
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:3219:49: error: passing argument 2 of ‘wmi_install_notify_handler’ from incompatible pointer type [-Wincompatible-pointer-types]
 3219 |                                                 acer_wmi_notify, NULL);
      |                                                 ^~~~~~~~~~~~~~~
      |                                                 |
      |                                                 void (*)(u32,  void *) {aka void (*)(unsigned int,  void *)}
In file included from ./include/linux/i2c.h:13,
                 from ./include/uapi/linux/fb.h:6,
                 from ./include/linux/fb.h:5,
                 from /home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:22:
./include/linux/acpi.h:404:60: note: expected ‘wmi_notify_handler’ {aka ‘void (*)(union acpi_object *, void *)’} but argument is of type ‘void (*)(u32,  void *)’ {aka ‘void (*)(unsigned int,  void *)’}
  404 |                                         wmi_notify_handler handler, void *data);
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: At top level:
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1977:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
 1977 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:3677:1: fatal error: opening dependency file /home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/.facer.o.d: Permission denied
 3677 | module_exit(acer_wmi_exit);
      | ^~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.build:229: /home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Error 1
make[2]: *** [/usr/lib/modules/6.12.1-arch1-1/build/Makefile:1936: /home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.12.1-arch1-1/build'
make: *** [Makefile:14: default] Error 2
📁 ~/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module
 ➥  sudo make
[sudo] password for reynard: 
make -C /lib/modules/6.12.1-arch1-1/build M=/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.12.1-arch1-1/build'
  CC [M]  /home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘acer_wmi_notify’:
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:2966:18: error: implicit declaration of function ‘wmi_get_event_data’ [-Wimplicit-function-declaration]
 2966 |         status = wmi_get_event_data(value, &response);
      |                  ^~~~~~~~~~~~~~~~~~
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: In function ‘acer_wmi_input_setup’:
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:3219:49: error: passing argument 2 of ‘wmi_install_notify_handler’ from incompatible pointer type [-Wincompatible-pointer-types]
 3219 |                                                 acer_wmi_notify, NULL);
      |                                                 ^~~~~~~~~~~~~~~
      |                                                 |
      |                                                 void (*)(u32,  void *) {aka void (*)(unsigned int,  void *)}
In file included from ./include/linux/i2c.h:13,
                 from ./include/uapi/linux/fb.h:6,
                 from ./include/linux/fb.h:5,
                 from /home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:22:
./include/linux/acpi.h:404:60: note: expected ‘wmi_notify_handler’ {aka ‘void (*)(union acpi_object *, void *)’} but argument is of type ‘void (*)(u32,  void *)’ {aka ‘void (*)(unsigned int,  void *)’}
  404 |                                         wmi_notify_handler handler, void *data);
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: At top level:
/home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1977:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
 1977 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:229: /home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Error 1
make[2]: *** [/usr/lib/modules/6.12.1-arch1-1/build/Makefile:1936: /home/reynard/Downloads/acer-predator-turbo-and-rgb-keyboard-linux-module] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.12.1-arch1-1/build'
make: *** [Makefile:14: default] Error 2

LarryTheMagicDragon, 2024-11-25:
I've seen a lot of errors like this since the update. #184 is the fix I'm using on my system. Try it and please let us know if it fixes your problem

Reynardd, 2024-11-25:
hello i tried it and it worked thanks :D 
i'll close this issue

closed 2024-11-25 by Reynardd