JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #209: Please add support for Predator Helios 18 (2024)

IBJamon, 2024-11-27:
# Model: `sudo dmidecode | grep "Product Name" -B 2 -A 4`
System Information
        Manufacturer: Acer
        Product Name: Predator PH18-72
        Version: V1.10
        Serial Number: NHQP4AA00141012E017600
        UUID: f36ef422-4927-3649-a89c-74d4dd4b3d58
        Wake-up Type: Power Switch
--
Base Board Information
        Manufacturer: RPL
        Product Name: IONIQ_RTX
        Version: V1.10
        Serial Number: NBQP41100241012E017600
        Asset Tag: Type2 - Board Asset Tag
        Features:

## Count of keyboard RGB zones: 0
I'm not sure, but I believe it's the same as the PH18-71

### Count of Cpu Fans: 1
### Count of Gpu Fans: 1  

### RGB keyboard works?  No
### Turbo button turn on fans?  No
### Turbo button turn on LED? No
### Turbo button activates overclock? No

Nothing happens for any of the above.  I tested with https://github.com/LarryTheMagicDragon/acer-predator-turbo-and-rgb-keyboard-linux-module.git fork because main does not yet support Arch's 6.12 kernels.

Please feel free to reach out of you need anything else!  My wife and I are both Linux users, so I would be happy to test anything you need.  I hope we can enable power boost overclock - that would be awesome, since Nvidia drivers don't allow this currently.

LarryTheMagicDragon, 2024-11-27:
@IBJamon I have done research on your model, and from what I can tell it uses per key RGB. The implementation we have support for only supports 4 zones of RGB, not per key RGB. Personally speaking I'm not sure I'll be able to add any kind of RGB for your model. I will do what I can to add support for your model where I can. I'm currently reading up on kernel programming so I can support this project better. I'll do as much as I can to help you.

can you run "sudo libinput debug-events" and tell me if anything shows up in the output when you press your turbo key?

IBJamon, 2024-11-27:
Unfortunately, neither the turbo button or the fancy 'Predator' button 
seem to trigger any events.

--------------rDoMJe064RsTuuDYZbS39uqO
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    Unfortunately, neither the turbo button or the fancy 'Predator'
    button seem to trigger any events.<br>
    <br>
    <br>
    <div class="moz-cite-prefix">On 11/27/24 11:41 AM,
      LarryTheMagicDragon wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto"><a class="user-mention notranslate" data-hovercard-type="user" data-hovercard-url="/users/IBJamon/hovercard" data-octo-click="hovercard-link-click" data-octo-dimensions="link_type:self" href="https://github.com/IBJamon" ***@***.***</a>
        I have done research on your model, and from what I can tell it
        uses per key RGB. The implementation we have support for only
        supports 4 zones of RGB, not per key RGB. Personally speaking
        I'm not sure I'll be able to add any kind of RGB for your model.
        I will do what I can to add support for your model where I can.
        I'm currently reading up on kernel programming so I can support
        this project better. I'll do as much as I can to help you.</p>
      <p dir="auto">can you run &quot;sudo libinput debug-events&quot; and tell me
        if anything shows up in the output when you press your turbo
        key?</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          ID: <span>&lt;JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/209/2504447251</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/209#issuecomment-2504447251",
"url": "https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/209#issuecomment-2504447251",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
    <br>
  </body>
</html>

--------------rDoMJe064RsTuuDYZbS39uqO--

LarryTheMagicDragon, 2024-11-27:
I'm not sure I'll be able to help in the next few days, but I do intend to help when I can. Do you have an email I can contact for this issue, or should I just respond to this thread?

LarryTheMagicDragon, 2024-11-27:
Thank you. I'll be looking into what can be done to add support for your model

IBJamon, 2024-11-27:
Thanks!  I may not be a very good programmer, but I'm a sysadmin, so whatever you need let me know :)

IBJamon, 2024-12-26:
Hey @LarryTheMagicDragon, we haven't talked in a while.  Is there anything I can do to help with this?

LarryTheMagicDragon, 2024-12-27:
When you run 'sudo libinput debug-events' do you see anything come up when you press the turbo button? That'll let us know if the button is recognized. As for the per key RGB I'm still looking into how to approach that.

IBJamon, 2024-12-27:
Unfortunately, neither the Turbo button or the Predator button respond to libinput.