JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #210: Installation problem

zChris83, 2024-11-30:
# Model: `sudo dmidecode | grep "Product Name" -B 2 -A 4`

I get this when i try to install it as a module for with sudo ./install.sh
`make -C /lib/modules/6.6.62-gentoo-dist/build M=/home/chris/Projekt/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Går till katalogen ”/usr/src/linux-6.6.62-gentoo-dist”
Makefile:2075: !!!! WARNING !!!!
Makefile:2076: This kernel was configured and installed by the package manager.
Makefile:2077: "make" should not be run manually here.
Makefile:2078: See also: https://wiki.gentoo.org/wiki/Project:Distribution_Kernel
Makefile:2079: See also: https://wiki.gentoo.org/wiki/Kernel/Configuration
Makefile:2080: !!!! WARNING !!!!
/usr/src/linux-6.6.62-gentoo-dist/Makefile:2075: !!!! WARNING !!!!
/usr/src/linux-6.6.62-gentoo-dist/Makefile:2076: This kernel was configured and installed by the package manager.
/usr/src/linux-6.6.62-gentoo-dist/Makefile:2077: "make" should not be run manually here.
/usr/src/linux-6.6.62-gentoo-dist/Makefile:2078: See also: https://wiki.gentoo.org/wiki/Project:Distribution_Kernel
/usr/src/linux-6.6.62-gentoo-dist/Makefile:2079: See also: https://wiki.gentoo.org/wiki/Kernel/Configuration
/usr/src/linux-6.6.62-gentoo-dist/Makefile:2080: !!!! WARNING !!!!
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-pc-linux-gnu-gcc (Gentoo 13.3.1_p20241025 p1) 13.3.1 20241024
  You are using:           gcc (Gentoo 14.2.1_p20241116 p3) 14.2.1 20241116
  CC [M]  /home/chris/Projekt/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/chris/Projekt/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: I funktion ”acer_wmi_input_setup”:
/home/chris/Projekt/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:3205:49: fel: skickar argument 2 till ”wmi_install_notify_handler” från inkompatibel pekartyp [-Wincompatible-pointer-types]
 3205 |                                                 acer_wmi_notify, NULL);
      |                                                 ^~~~~~~~~~~~~~~
      |                                                 |
      |                                                 void (*)(union acpi_object *, void *)
I filen inkluderad ifrån ./include/linux/i2c.h:13,
                   ifrån ./include/uapi/linux/fb.h:6,
                   ifrån ./include/linux/fb.h:7,
                   ifrån /home/chris/Projekt/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:22:
./include/linux/acpi.h:415:60: anm: ”wmi_notify_handler” {även ”void (*)(unsigned int,  void *)”} förväntades men argumentet har typ ”void (*)(union acpi_object *, void *)”
  415 |                                         wmi_notify_handler handler, void *data);
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~
/home/chris/Projekt/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c: På toppnivå:
/home/chris/Projekt/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1977:20: varning: ”WMID_gaming_get_u64” är definierad men inte använd [-Wunused-function]
 1977 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:243: /home/chris/Projekt/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o] Fel 1
make[2]: *** [/usr/src/linux-6.6.62-gentoo-dist/Makefile:1929: /home/chris/Projekt/acer-predator-turbo-and-rgb-keyboard-linux-module] Fel 2
make[1]: *** [Makefile:234: __sub-make] Fel 2
make[1]: Lämnar katalogen ”/usr/src/linux-6.6.62-gentoo-dist”
make: *** [Makefile:14: default] Fel 2
rmmod: ERROR: Module acer_wmi is not currently loaded
rmmod: ERROR: Module facer is not currently loaded
insmod: ERROR: could not load module src/facer.ko: No such file or directory
[ 1133.205479] input: Logitech G Pro Wireless Gaming Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:046D:C088.0026/input/input60
[ 1133.327668] logitech-hidpp-device 0003:046D:C088.0026: input,hidraw10: USB HID v1.11 Keyboard [Logitech G Pro Wireless Gaming Mouse] on usb-0000:00:14.0-3/input1
[ 1133.389177] logitech-hidpp-device 0003:046D:C088.0027: HID++ 4.2 device connected.
[ 1133.417699] logitech-hidpp-device 0003:046D:C088.0027: hiddev101,hidraw11: USB HID v1.11 Device [Logitech G Pro Wireless Gaming Mouse] on usb-0000:00:14.0-3/input2
[ 1133.555701] elogind-daemon[2281]: Watching system buttons on /dev/input/event26 (Logitech G Pro Wireless Gaming Mouse)
[ 1136.913436] usb usb1-port3: disabled by hub (EMI?), re-enabling...
[ 1136.913458] usb 1-3: USB disconnect, device number 15
[ 3980.789190] acer_wmi: Acer Laptop WMI Extras unloaded
[ 3980.887166] facer: version magic '6.4.14-gentoo-dist SMP preempt mod_unload ' should be '6.6.62-gentoo-dist SMP preempt mod_unload '
[ 4230.519254] facer: version magic '6.4.14-gentoo-dist SMP preempt mod_unload ' should be '6.6.62-gentoo-dist SMP preempt mod_unload '
[*] Done
`

yhanruzai, 2024-12-01:
Can confirm this error still happens when trying to build it. Running on NixOS, but this shouldn't matter.

HollowMan6, 2024-12-01:
Should be fixed in PR #211

zChris83, 2024-12-01:
It compiled with your changes. My next problem is:

insmod: ERROR: could not insert module src/facer.ko: Unknown symbol in module
[  217.874263] wlp0s20f3: authenticate with 04:92:26:45:e9:24
[  217.879557] wlp0s20f3: send auth to 04:92:26:45:e9:24 (try 1/3)
[  217.910698] wlp0s20f3: authenticated
[  217.914104] wlp0s20f3: associate with 04:92:26:45:e9:24 (try 1/3)
[  217.918451] wlp0s20f3: RX AssocResp from 04:92:26:45:e9:24 (capab=0x911 status=0 aid=3)
[  217.925123] wlp0s20f3: associated
[  217.947387] wlp0s20f3: Limiting TX power to 30 (30 - 0) dBm as advertised by 04:92:26:45:e9:24
[  359.160110] facer: Unknown symbol platform_profile_register (err -2)
[  359.160242] facer: Unknown symbol platform_profile_notify (err -2)
[  359.160463] facer: Unknown symbol platform_profile_remove (err -2)
[*] Done

HollowMan6, 2024-12-01:
I didn't encounter a related error like yours on my side, but from the error logs, it looks like you didn't enable the kernel configuration option `CONFIG_ACPI_PLATFORM_PROFILE`, and I guess you will need to reconfigure and rebuild your kernel with `CONFIG_ACPI_PLATFORM_PROFILE` enabled.

zChris83, 2024-12-01:
Its not a LTS kernel. uname -r gives 6.6.62-gentoo-dist. However i did a modprobe platform_profile and that seems to get rid of the error message when insmod facer. Although python keyboard.py doesnt affect the rgbs on the keyboard.

yhanruzai, 2024-12-01:
So, it turns out a combination of building with @LarryTheMagicDragon build and changing the kernel from LTS to the latest made it somehow work.

zChris83, 2024-12-01:
> So, it turns out a combination of building with @LarryTheMagicDragon build and changing the kernel from LTS to the latest made it somehow work.

Were you able to change the RGB on your keyboard for PH18-71 ?

yhanruzai, 2024-12-01:
So update, it seems the fix depends on updating the kernel away from LTS, which causes my WM to flicker unless I use the stable nvidia drivers which needs to downgrade the kernel making the module to break again.

LarryTheMagicDragon, 2024-12-01:
@yhanruzai I'm a little busy at the moment, but I have added fixing LTS to my todo list.

PR #211 (2024-12-13): acer_wmi_notify should still support kernel version less than 6.12
Will need some review work (confirmation) from people using kernel version 6.12 as I'm not using it.

Related to #184

Fixes #210 

Error now when compiled in kernel version 6.11:

```log
./src/facer.c: In function ‘acer_wmi_input_setup’: ./src/facer.c:3205:49: error: passing argument 2 of ‘wmi_install_notify_handler’ from incompatible pointer type [-Wincompatible-pointer-types]
 3205 |                                                 acer_wmi_notify, NULL);
      |                                                 ^~~~~~~~~~~~~~~ [... cut]
closed 2024-12-13 by mmsaeed509