JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #218: PH16-71 DOES NOTHING

veryfunnyaz, 2024-12-30:
Well, I have the service up and running and every time I try to apply something it does nothing, I tried to use the GUI to have a more clear view and apparently its saying the payload is being written and the setting applied while they arent. Nothing works. Could you add support for this version?

|Feature|Value|
|---	|---	|
|Model|Predator PH16-71|
|Number of CPU Fans| ? |
|Number of GPU Fans| ? |
|Number of RGB Zones| ? |
|RGB keyboard works?|  No |
|Turbo button turns on fans?| No |
|Turbo button turn on LED?| No |
|Turbo button activates overclock?| No |
|Distribution Specific Information|EndeavourOS Linux|
|Kernel Version|Linux eos 6.12.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 27 Dec 2024 14:24:37 +0000 x86_64 GNU/Linux|

[image]

abruenin, 2024-12-30:
How did you get that GUI?

veryfunnyaz, 2024-12-30:
https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module?tab=readme-ov-file#roadmap

its the one that says GUI(wxPython with tray icon): ([x211321](https://github.com/x211321/RGB-Config-Acer-gkbbl-0))

veryfunnyaz, 2024-12-30:
> How did you get that GUI?

https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module?tab=readme-ov-file#roadmap

its the one that says GUI(wxPython with tray icon): ([x211321](https://github.com/x211321/RGB-Config-Acer-gkbbl-0))

x211321, 2024-12-30:
When the GUI states that the payload has been written successfully, we can at least assume that the kernel module is correctly installed and loaded. Otherwise you would get a message that the RGB device is not available.

If you see no change on your keyboard it probably means that your model is not supported by the kernel model.

I might not be up-to-date on the current state of the kernel module, but as far as I know only 4 zone RGB keyboards are currently supported. Your model appears to have individually addressable RGB keys instead.

abruenin, 2025-01-01:
> its the one that says GUI(wxPython with tray icon): ([x211321](https://github.com/x211321/RGB-Config-Acer-gkbbl-0))

Ah thanks, the GUI works, but still no reaction from my keyboard.

> I might not be up-to-date on the current state of the kernel module, but as far as I know only 4 zone RGB keyboards are currently supported. Your model appears to have individually addressable RGB keys instead.

Hm, looks like you are right. Haven't double checked because I read that it works here:
[https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/175](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/175)

Maybe this is what 

> Implement RGB Dynamic effects (per key RGB)
> Implement RGB Static coloring (per key RGB)

from the roadmap means?

veryfunnyaz, 2025-01-01:
> Hm, looks like you are right. Haven't double checked because I read that it works here: #175

thats clearly not the same model. it says PHN16-71 not PH16-71

closed 2025-03-09 by veryfunnyaz