JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #219: Turbo Mode Not Working [ Predator PHN16-71 ]

kumarvivek1752, 2025-01-01:
## Is this an issue or a reproduction result?  ✔️ OR ⁉️
## Please fill in the template 
|Feature|Value|
|---	|---	|
|Model|```Predator PHN16-71```|
|Number of CPU Fans| 1 |
|Number of GPU Fans| 1 |
|Number of RGB Zones| 4|
|RGB keyboard works?|  Yes |
|Turbo button turns on fans?| No |
|Turbo button turn on LED?| No |
|Turbo button activates overclock?| No |
|Distribution Specific Information|```  Distributor ID: Arch Description:    Arch Linux Release:        rolling  ```|
|Kernel Version|`Linux archlinux 6.12.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 27 Dec 2024 14:24:37 +0000 x86_64 GNU/Linux`|

RGB Keyboard working fine, but turbo not working

logs:
```
➜  turbo-fan sudo systemctl start turbo-fan.service
➜  turbo-fan sudo systemctl status turbo-fan.service
× turbo-fan.service - Enables turbo button
     Loaded: loaded (/etc/systemd/system/turbo-fan.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Wed 2025-01-01 22:27:09 IST; 1s ago
   Duration: 86ms
 Invocation: 8e6bd518a57d4e0ca38f675452da076f
    Process: 4626 ExecStart=/bin/bash /opt/turbo-fan/service.sh (code=exited, status=139)
   Main PID: 4626 (code=exited, status=139)
   Mem peak: 2.7M
        CPU: 36ms

Jan 01 22:27:09 archlinux systemd[1]: Started Enables turbo button.
Jan 01 22:27:09 archlinux bash[4635]: rmmod: ERROR: Module acer_wmi is not currently loaded
Jan 01 22:27:09 archlinux bash[4636]: rmmod: ERROR: Module facer is in use
Jan 01 22:27:09 archlinux bash[4626]: /opt/turbo-fan/service.sh: line 17:  4640 Segmentation fault      insmod src/facer.ko
Jan 01 22:27:09 archlinux systemd[1]: turbo-fan.service: Main process exited, code=exited, status=139/n/a
Jan 01 22:27:09 archlinux systemd[1]: turbo-fan.service: Failed with result 'exit-code'.
➜  turbo-fan

```

Arnesfield, 2025-01-06:
Same here for **PH315-54**, turbo does not work with the latest updates.

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.1 LTS
Release:	24.04
Codename:	noble
```

```
$ uname -a
Linux aurora 6.8.0-51-generic #52-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec  5 13:09:44 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

```
$ sudo ./install_service.sh 
[Mode: install]
[Remove old data]
[Create directories]
[Copy new data]
sending incremental file list
LICENSE
Makefile
Module.symvers
README.md
dkms.conf
facer_rgb.py
install.sh
install_openrc.sh
install_service.sh
keyboard.py
keyboard.webp
modules.order
push.sh
refresh.sh
service.sh
uninstall.sh
uninstall_service.sh
PKGBUILD/
PKGBUILD/PKGBUILD
PKGBUILD/Predator-Sense-systemd-git.install
PKGBUILD/makepkg.sh
src/
src/.facer.ko.cmd
src/.facer.mod.cmd
src/.facer.mod.o.cmd
src/.facer.o.cmd
src/facer.c
src/facer.ko
src/facer.mod
src/facer.mod.c
src/facer.mod.o
src/facer.o

sent 3,967,505 bytes  received 602 bytes  7,936,214.00 bytes/sec
total size is 3,964,396  speedup is 1.00
[Create turbo-fan service]
Synchronizing state of turbo-fan.service with SysV service script with /usr/lib/systemd/systemd-sysv-install.
Executing: /usr/lib/systemd/systemd-sysv-install enable turbo-fan
update-rc.d: error: turbo-fan Default-Start contains no runlevels, aborting.
```

For a while now (when I had the `6.8.0-49-generic` kernel), I've stuck with the commit abc8a072d0024c75e8d5271363f7ab706e8562dc, since it was the last commit before turbo stopped working. Starting at commit 1b99bc8876eff8b253251ed5e1171ea091bfb3c4 and turbo does not work, at least for my kernel version.

0x7375646F, 2025-01-11:
i've tested this module and default kernel module named acer-wmi the acer-wmi works without any problem.
well the only downside is there is no any keyboard rgb implemented there but turbo key does work.

kumarvivek1752, 2025-01-11:
> i've tested this module and default kernel module named acer-wmi the acer-wmi works without any problem. well the only downside is there is no any keyboard rgb implemented there but turbo key does work.

How are you controlling both your CPU and GPU fans?

0x7375646F, 2025-01-11:
The fan control mechanism is not implemented too
Iam working on similar project where iam integrating all the features in it might take some time but i have finished implementation for lcd override now next is fan speeds and then rewrite of rgb keyboard mechanisms , battery limiter and so on. 
Will update yall if i complete it in time.

kumarvivek1752, 2025-01-11:
if it is open source I'll be happy to contribute to your project

0x7375646F, 2025-01-11:
It indeed will be open source rn the source code is kinda messy so once i implement custom fan speed feature i can send it to you to test. if you have discord you can contact me.

kumarvivek1752, 2025-01-11:
@0x7375646F, I only need fan control because when I run GPU-intensive tasks, the temperature of my laptop rises to around 95°C–98°C, but the fans still run at medium speed.

This is my Discord ID: vivek1752
If you allow me on your project, I can try to implement this feature myself.

0x7375646F, 2025-01-11:
@kumarvivek1752  the fan speed control will be implemented by tomorrow accept my friend request in dc its sudo.dwg

tribhuwan-kumar, 2025-01-11:
> @kumarvivek1752  the fan speed control will be implemented by tomorrow accept my friend request in dc its sudo.dwg

@kumarvivek1752 @0x7375646F can I collaborate too?

0x7375646F, 2025-01-11:
@tribhuwan-kumar  sure well i haven't made any repo but i have implemented custom fan speed 
https://streamable.com/z7o66x
iam thinking of making repo once all the features are done  and the code is properly cleaned but lets see hmm.
if you need code for testing you can send friend request to my discord acc.

kumarvivek1752, 2025-01-11:
@0x7375646F I tested your code, and it is working as expected.

rishi-2406, 2025-02-07:
I am having a similar issue where after installing , i am not able to use my turbo button. Is there a fix ?

> [@0x7375646F](https://github.com/0x7375646F) I tested your code, and it is working as expected.

Can i get it as well

0x7375646F, 2025-02-07:
> I am having a similar issue where after installing , i am not able to use my turbo button. Is there a fix ?
> 
> > [@0x7375646F](https://github.com/0x7375646F) I tested your code, and it is working as expected.
> 
> Can i get it as well

https://github.com/0x7375646F/Linuwu-Sense