JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #220: How does WMID_gaming_set_fan_mode() work

Wer-Wolf, 2025-01-06:
Hello,

i am currently try to add support for manual fan control to the acer-wmi driver. Setting the fan speed already works, but i need to find a way to switch the fans back to automatic mode.

Because of this i want to ask how `WMID_gaming_set_fan_mode()` works internally.

Thanks,
Armin Wolf

JafarAkhondali, 2025-01-07:
Hello,
What happens when you use this?
```

		turbo_state = 0;
		/* Turn off turbo led */
		WMID_gaming_set_u64(0x1, ACER_CAP_TURBO_LED);

		/* Set FAN mode to auto */
		WMID_gaming_set_fan_mode(0x1);

		/* Set OC to normal */
		WMID_gaming_set_u64(0x5, ACER_CAP_TURBO_OC);
		WMID_gaming_set_u64(0x7, ACER_CAP_TURBO_OC);
```

Wer-Wolf, 2025-01-07:
This works, but i want to control the fan mode for each fan separately.

snowyoneill, 2025-01-21:
@Wer-Wolf - you should check out this repo: https://github.com/snowyoneill/Linux-PredatorSense

Wer-Wolf, 2025-01-21:
Interesting, but i i am using the WMI interface.

Anyway i will try to perform some tests myself.

Wer-Wolf, 2025-02-03:
Thank you very much, this is exactly what i need.

JafarAkhondali, 2025-02-03:
@Wer-Wolf Thanks to @0x7375646F :)

By any chance @Wer-Wolf, I see you also work on Dell Alienware, It looks like the M16 R2 model is using USB for RGB control & other things, but I couldn't find any code implemented in Linux kernel. Do you know if it's implemented or if there has been some progress on it? If not, I might be able to help.

Wer-Wolf, 2025-02-03:
I do not directly work on alienware devices, but there is another developer working on it right now.

If you want i can give him your contact info.

kuu-rt, 2025-02-04:
Hi Jafar,

> By any chance [@Wer-Wolf](https://github.com/Wer-Wolf), I see you also work on Dell Alienware, It looks like the M16 R2 model is using USB for RGB control & other things (not WMI), but I couldn't find any code implemented in the Linux kernel. Do you know if it's implemented or if there has been some progress on it? If not, I might be able to help.

I am currently working on the alienware-wmi upstream linux driver. 

As you mentioned, RGB control is not done through a WMI interface, so unfortunately it lies out of my knowledge. Also this USB device is present on most newer (maybe >2016? I could be wrong) Alienware and Dell gaming laptops. This is the device present on my laptop Alienware x15 R1:
```
Bus 003 Device 002: ID 187c:0550 Alienware Corporation LED controller
```
Thankfully no kernel drivers are required, because these devices are already exposed through the HIDRAW interface, but as I said, I don't have experience with this so I don't know for sure. 

It would be amazing to have a tool for RGB control and probably contributing to OpenRGB is the way to go.

This project could also be of great help [alienfx-tools](https://github.com/T-Troll/alienfx-tools). It supports A LOT of models, has some documentation and the developer is very helpful and active.

If you end up working on this let me know, I'd love to help in some way!

JafarAkhondali, 2025-02-04:
Hi @kuu-rt 
I can confirm that the model I'm working with (90H8X54 - M16 R2) uses USB. I've seen some WMI requests for thermal control, but I have not yet checked that part carefully.
Interestingly, the USB device ID is similar but not the same **(notice that last digit)**:
```Bus 003 Device 005: ID 187c:0551 Alienware Corporation AW-ELC```.

I was able to change the touchpad color in Linux using USB. If I can have some better progress I'll create a repo for it later. Contributing to OpenRGB would be nice, however, I still prefer some features to be available in the Linux kernel by default, without 3rd party software. I've also tried using alien fx-tools but it didn't work for this laptop on Windows. The only issue is that there are TONS of features for this laptop:

1. Per key RGB
2. Touchpad RGB
3. Alien head logo in front
4. Alien head logo back
5. quiet mode, high-performance mode, etc

Then for each coloring mode, there are many different modes as well(static coloring, dynamic coloring ...)