JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #221: I was writed a driver to control fan speeds.

DetuxTR, 2025-01-19:
Hello guys, i was completed driver to control fan speeds on Acer Nitro/Predator. Now i'm working on daemon. If you wanting to give a try to it: https://github.com/DetuxTR/AcerNitroLinuxGamingDriver. Otherwise, the project needs the support of community.

snowyoneill, 2025-01-21:
@DetuxTR check out - https://github.com/snowyoneill/Linux-PredatorSense

DetuxTR, 2025-01-21:
The ec method is not working on 16 series # machines :(. But your project looks very good.

snowyoneill, 2025-01-21:
@DetuxTR ohhh really!! Did Acer remove it? Do you have an example where it doesnt work? I can try and take a look into updating it

snowyoneill, 2025-01-21:
@DetuxTR yeah ill check it out and report back.

I would confirm that the EC access has been removed by using [RWEverything – Read & Write Everything](https://rweverything.com/) on Windows. If you dont see the hex values change ( the table of associated hex values to Predator functionality are in my main.py file ) then your driver will be the primary way of going. Otherwise it's still alive :)

DetuxTR, 2025-01-21:
Thanks for your attention, you can ensure that i was spended my hours by watching the ec table like cypher when watching the matrix. But i will give it a second change, and i will try your solution. I will report back too.> @DetuxTR yeah ill check it out and report back.
> 
> I would confirm that the EC access has been removed by using [RWEverything – Read & Write Everything](https://rweverything.com/) on Windows. If you dont see the hex values change as you toggle settings in PredatorSense ( the table of associated hex values to Predator functionality are in my main.py file ) then your driver will be the primary way of going. Otherwise it's still alive :)