JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #225: Cannot install as service on acer phn16 72.

wammmmmmm, 2025-01-29:
Cannot install program as service. It works when using sudo ./install.sh
Using: 
Fedora 41
Acer phn16-72

This is what I get after using sudo ./install_service.sh

[Mode: install]
['turbo-fan' service is presented. Remove it.]
Synchronizing state of turbo-fan.service with SysV service script with /usr/lib/systemd/systemd-sysv-install.
Executing: /usr/lib/systemd/systemd-sysv-install disable turbo-fan
service turbo-fan does not support chkconfig
[Remove old data]
removed '/opt/turbo-fan/PKGBUILD/PKGBUILD'
removed '/opt/turbo-fan/PKGBUILD/Predator-Sense-systemd-git.install'
removed '/opt/turbo-fan/PKGBUILD/makepkg.sh'
removed directory '/opt/turbo-fan/PKGBUILD'
removed '/opt/turbo-fan/src/facer.c'
removed directory '/opt/turbo-fan/src'
removed '/opt/turbo-fan/LICENSE'
removed '/opt/turbo-fan/Makefile'
removed '/opt/turbo-fan/README.md'
removed '/opt/turbo-fan/dkms.conf'
removed '/opt/turbo-fan/facer_rgb.py'
removed '/opt/turbo-fan/install.sh'
removed '/opt/turbo-fan/install_openrc.sh'
removed '/opt/turbo-fan/install_service.sh'
removed '/opt/turbo-fan/keyboard.py'
removed '/opt/turbo-fan/keyboard.webp'
removed '/opt/turbo-fan/push.sh'
removed '/opt/turbo-fan/refresh.sh'
removed '/opt/turbo-fan/uninstall.sh'
removed '/opt/turbo-fan/uninstall_service.sh'
removed directory '/opt/turbo-fan'
[Create directories]
[Copy new data]
sending incremental file list
LICENSE
Makefile
README.md
dkms.conf
facer_rgb.py
install.sh
install_openrc.sh
install_service.sh
keyboard.py
keyboard.webp
push.sh
refresh.sh
uninstall.sh
uninstall_service.sh
PKGBUILD/
PKGBUILD/PKGBUILD
PKGBUILD/Predator-Sense-systemd-git.install
PKGBUILD/makepkg.sh
src/
src/facer.c

sent 2,680,666 bytes  received 374 bytes  5,362,080.00 bytes/sec
total size is 2,678,725  speedup is 1.00
[Create turbo-fan service]
chmod: cannot access '/opt/turbo-fan/service.sh': No such file or directory
Synchronizing state of turbo-fan.service with SysV service script with /usr/lib/systemd/systemd-sysv-install.
Executing: /usr/lib/systemd/systemd-sysv-install enable turbo-fan
service turbo-fan does not support chkconfig

LarryTheMagicDragon, 2025-01-29:
@wammmmmmm I have created a patch #226. Can you tell me if it works?

wammmmmmm, 2025-01-30:
I replaced the original file with your patch and then ran it with sudo. I got this: 

[Mode: install]
['turbo-fan' service is presented. Remove it.]
Synchronizing state of turbo-fan.service with SysV service script with /usr/lib/systemd/systemd-sysv-install.
Executing: /usr/lib/systemd/systemd-sysv-install disable turbo-fan
service turbo-fan does not support chkconfig
[Remove old data]
removed '/opt/turbo-fan/PKGBUILD/PKGBUILD'
removed '/opt/turbo-fan/PKGBUILD/Predator-Sense-systemd-git.install'
removed '/opt/turbo-fan/PKGBUILD/makepkg.sh'
removed directory '/opt/turbo-fan/PKGBUILD'
removed '/opt/turbo-fan/src/facer.c'
removed '/opt/turbo-fan/src/facer.mod'
removed '/opt/turbo-fan/src/facer.o'
removed '/opt/turbo-fan/src/.facer.o.cmd'
removed '/opt/turbo-fan/src/.facer.mod.cmd'
removed '/opt/turbo-fan/src/facer.mod.c'
removed '/opt/turbo-fan/src/facer.mod.o'
removed '/opt/turbo-fan/src/.facer.mod.o.cmd'
removed '/opt/turbo-fan/src/facer.ko'
removed '/opt/turbo-fan/src/.facer.ko.cmd'
removed directory '/opt/turbo-fan/src'
removed '/opt/turbo-fan/LICENSE'
removed '/opt/turbo-fan/Makefile'
removed '/opt/turbo-fan/README.md'
removed '/opt/turbo-fan/dkms.conf'
removed '/opt/turbo-fan/facer_rgb.py'
removed '/opt/turbo-fan/install.sh'
removed '/opt/turbo-fan/install_openrc.sh'
removed '/opt/turbo-fan/install_service.sh'
removed '/opt/turbo-fan/keyboard.py'
removed '/opt/turbo-fan/keyboard.webp'
removed '/opt/turbo-fan/push.sh'
removed '/opt/turbo-fan/refresh.sh'
removed '/opt/turbo-fan/uninstall.sh'
removed '/opt/turbo-fan/uninstall_service.sh'
removed '/opt/turbo-fan/.modules.order.cmd'
removed '/opt/turbo-fan/.Module.symvers.cmd'
removed '/opt/turbo-fan/.module-common.o'
removed '/opt/turbo-fan/..module-common.o.cmd'
removed '/opt/turbo-fan/modules.order'
removed '/opt/turbo-fan/Module.symvers'
removed '/opt/turbo-fan/service.sh'
removed directory '/opt/turbo-fan'
[Create directories]
[Copy new data]
sending incremental file list
LICENSE
Makefile
Module.symvers
README.md
dkms.conf
facer_rgb.py
install.sh
install_openrc.sh
install_service.sh
keyboard.py
keyboard.webp
modules.order
push.sh
refresh.sh
service.sh
uninstall.sh
uninstall_service.sh
PKGBUILD/
PKGBUILD/PKGBUILD
PKGBUILD/Predator-Sense-systemd-git.install
PKGBUILD/makepkg.sh
src/
src/.facer.ko.cmd
src/.facer.mod.cmd
src/.facer.mod.o.cmd
src/.facer.o.cmd
src/facer.c
src/facer.ko
src/facer.mod
src/facer.mod.c
src/facer.mod.o
src/facer.o

sent 4,320,515 bytes  received 602 bytes  8,642,234.00 bytes/sec
total size is 4,317,363  speedup is 1.00
[Create turbo-fan service]
Synchronizing state of turbo-fan.service with SysV service script with /usr/lib/systemd/systemd-sysv-install.
Executing: /usr/lib/systemd/systemd-sysv-install enable turbo-fan
service turbo-fan does not support chkconfig

Seems to be the same message, unless Iḿ missing something.

LarryTheMagicDragon, 2025-01-30:
@wammmmmmm this is a new message, the old error isn't there. I'll look into how to fix this new problem

LarryTheMagicDragon, 2025-02-02:
@wammmmmmm I just want to let you know I haven't forgotten about this. As far as I can tell the issue here is the "service turbo-fan does not support chkconfig" message. I've never seen or heard of chkconfig before today, so I'm having a harder time to fix this.

Also you said "I'm also using cachyos (as my kernel)" do you mean you're using the CachyOS kernel as a custom kernel on another distro, or are you using CachyOS?

wammmmmmm, 2025-02-04:
Thank you for not forgetting!! I'm not very knowledgeable in programming YET, but if there's anything that you'd like me to do to help, I'd be more than welcome. 

And yes, I was using CachyOS as a kernel in fedora 41. I'm thinking of switching to arch, with the linux-zen kernel. That might rule out the possibility of the kernel being the problem. I'll update you once I change.

wammmmmmm, 2025-03-03:
Hello again brothers. I did not forget to update, I just have been busy with work and until now I am able to post. So, what changed on my system: I switched to arch. I tried to install this first with the linux zen kernel, and then I switched to the cachyos eevdf kernel. I couldn't install as service with both. I tried installing through the exodia os repo, as indicated in the github readme, but I always got this error: 

resolving dependencies...
looking for conflicting packages...

Packages (2) linux-headers-6.13.5.arch1-1  Predator-Sense-CLI-8.0-3

Total Download Size:    27.31 MiB
Total Installed Size:  147.11 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 linux-headers-6.13.5.arch1...    27.3 MiB  2.97 MiB/s 00:09 [---------------------------------] 100%
 Predator-Sense-CLI-8.0-3-any is up to date
 Total (2/2)                      27.3 MiB  2.94 MiB/s 00:09 [---------------------------------] 100%
error: failed retrieving file 'Predator-Sense-CLI-8.0-3-any.pkg.tar.zst.sig' from exodia-os.github.io : The requested URL returned error: 404
warning: failed to retrieve some files
error: failed to commit transaction (failed to retrieve some files)
Errors occurred, no packages were upgraded.
 -> error installing repo packages

So this entire time I have been launching the rgb app with install.sh every time I turn on my computer. BUT it seems that it works now. I don't know what happened. I tried install_service.sh last week and it wasn't working. I have not installed anything or done anything to my system except for a full system update on saturday (sudo pacman -Syu) and while writing this I ran install_service.sh to get the error message and paste it here but I did not get the error. It just worked. My theory is that maybe arch added something for compatibility with my laptop (?). Maybe, idk. 

Well. that's it. I'm closing the issue. Still I want to thank my boy Larry for his time. Love you brother thanks for helping, I really appreciate it.

closed 2025-03-03 by wammmmmmm

LarryTheMagicDragon, 2025-03-03:
@wammmmmmm were you able to try #226?

commit 66464ea9 (2025-05-06): This should fix #225
commit ab922571 (2025-05-06): Update install_service.sh

Fixed #225 issue