JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #230: ERROR: Module acer_wmi is not currently loaded

patrickalima98, 2025-03-23:
## Is this an issue or a reproduction result?  ✔️ OR ⁉️
## Please fill in the template 
|Feature|Value|
|---	|---	|
|Model|```sudo dmidecode \| grep "Product Name" -B 2 -A 4```|
|Number of CPU Fans| 1 |
|Number of GPU Fans| 1 |
|Number of RGB Zones| PerKeyRGB|
|RGB keyboard works?|  No |
|Turbo button turns on fans?| No |
|Turbo button turn on LED?| No |
|Turbo button activates overclock?| No |
|Kernel Version|`inux helios16 6.12.10-76061203-generic #202412060638~1740154617~22.04~b4b3ebc SMP PREEMPT_DYNAMIC Fri F x86_64 x86_64 x86_64 GNU/Linux`|

This is my output:

Make:

```
make -C /lib/modules/6.12.10-76061203-generic/build M=/home/patrickalima98/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.10-76061203-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  CC [M]  /home/patrickalima98/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.o
/home/patrickalima98/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.c:1998:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
 1998 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
  MODPOST /home/patrickalima98/acer-predator-turbo-and-rgb-keyboard-linux-module/Module.symvers
  CC [M]  /home/patrickalima98/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.mod.o
  LD [M]  /home/patrickalima98/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
  BTF [M] /home/patrickalima98/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko
Skipping BTF generation for /home/patrickalima98/acer-predator-turbo-and-rgb-keyboard-linux-module/src/facer.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.12.10-76061203-generic'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi

```

Install:
```
make -C /lib/modules/6.12.10-76061203-generic/build M=/home/patrickalima98/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.10-76061203-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
make[1]: Leaving directory '/usr/src/linux-headers-6.12.10-76061203-generic'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
rmmod: ERROR: Module acer_wmi is not currently loaded
rmmod: ERROR: Module facer is in use

The dmesg log:
```
[  322.732307]  acer_wmi_init+0x6aa/0x1d90 [facer]
[  322.732313]  ? __pfx_acer_wmi_init+0x10/0x10 [facer]
[  322.732429] RIP: 0010:acer_wmi_init+0x6b8/0x1d90 [facer]
[  322.732482]  ? acer_wmi_init+0x6b8/0x1d90 [facer]
[  322.732486]  ? acer_wmi_init+0x6aa/0x1d90 [facer]
[  322.732491]  ? __pfx_acer_wmi_init+0x10/0x10 [facer]
[  323.070047] RIP: 0010:acer_wmi_init+0x6b8/0x1d90 [facer]
[  985.267238] acer_wmi: Acer Laptop ACPI-WMI Extras
[  985.267270] acer_wmi: Function bitmap for Communication Button: 0x801
[ 1208.676002] acer_wmi: Acer Laptop ACPI-WMI Extras
[ 1208.676073] acer_wmi: Function bitmap for Communication Button: 0x801
[ 1907.434042] acer_wmi: Acer Laptop ACPI-WMI Extras
[ 1907.434079] acer_wmi: Function bitmap for Communication Button: 0x801

``

My laptop: `Predator PH16-71`

Can you help me?

maxie7, 2025-03-28:
I have the same issue:

`Predator PHN16-71`

```
sudo ./install.sh                                                                                              ✔  00:00:23  
make -C /lib/modules/6.11.0-21-generic/build M=/home/maxie/src/ascripts/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-headers-6.11.0-21-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
make[1]: Leaving directory '/usr/src/linux-headers-6.11.0-21-generic'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
rmmod: ERROR: Module acer_wmi is not currently loaded
rmmod: ERROR: Module facer is not currently loaded
insmod: ERROR: could not insert module src/facer.ko: Invalid parameters
[  879.437613] facer: disagrees about version of symbol devm_hwmon_device_register_with_info
[  879.437615] facer: Unknown symbol devm_hwmon_device_register_with_info (err -22)
[  879.437627] facer: disagrees about version of symbol param_ops_bool
[  879.437629] facer: Unknown symbol param_ops_bool (err -22)
[  879.437633] facer: disagrees about version of symbol device_destroy
[  879.437634] facer: Unknown symbol device_destroy (err -22)
[  879.437677] facer: disagrees about version of symbol param_ops_int
[  879.437678] facer: Unknown symbol param_ops_int (err -22)
[  879.437681] facer: disagrees about version of symbol debugfs_create_dir
[  879.437682] facer: Unknown symbol debugfs_create_dir (err -22)
[*] Done
```

maxie7, 2025-03-29:
The problem is solved temporarily by adding execution of a script to root crontab.

(1) `sudo crontab -e -u root`

(2) add this line to crontab: `@reboot sudo /home/...path_to_script..././statrup_script.sh

(3)  the "startup_script.sh" listing:

```
#!/bin/zsh

# Navigate to the directory
cd /home/...path_to_cloned_repo_/acer-predator-turbo-and-rgb-keyboard-linux-module

# Run the installation script with sudo
./install.sh

# Run the RGB script
./facer_rgb.py -m 0 -z 1 -s 0 -b 40 -cR 255 -cB 255 -cG 0
./facer_rgb.py -m 0 -z 2 -s 0 -b 40 -cR 255 -cB 255 -cG 0
./facer_rgb.py -m 0 -z 3 -s 0 -b 40 -cR 255 -cB 255 -cG 0
./facer_rgb.py -m 0 -z 4 -s 0 -b 40 -cR 255 -cB 0 -cG 255
```

(4) test it: reboot your laptop