JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #232: Acer Predator PHN14-51

Todd0042, 2025-03-30:
RGB working on CachyOS with 6.12 LTS Kernel - controlled via python script.
Turbo Unknown - no apparent change.
Unable to install on CachyOS with up to date 6.14 Kernel.

❯ sudo dmidecode | grep "Product Name" -B 2 -A 4

System Information
	Manufacturer: Acer
	Product Name: Predator PHN14-51
	Version: V1.01
	Serial Number: NHQRNAA00141215DF83400
	UUID: d2565073-e8e4-ee11-80f2-40c2baf2e7d5
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: MTL
	Product Name: Taycan_MTH
	Version: V1.01
	Serial Number: NBQQM11004412D64A13400
	Asset Tag:
	Features:

----------------------------  dsdt.aml ---------------------------
#!/bin/bash
# This script can install or uninstall acer turbo fan service. It means, that your turbo fan button still will be available even after rebooting.
# To install service just run this script as the root (sudo) user.
# After installation you can manage it as a usual service manually. Example: 'systemctl start/stop turbo-fan',  'systemctl enable/disable turbo-fan'
# To uninstall service, run this script with 'remove' argument. Example: 'sudo bash ./install_service.sh remove'.
# Note!!! Before removing, don't forget to switch off the turbo button because you will have forever turbo fan :)
mode=${1:-install} # Allowed modes: "install" and "remove". Default: install.
service=turbo-fan # Service name
target_dir=/opt/turbo-fan # Instalation folder
service_dir=/etc/systemd/system # Service setup folder (where all services are stored)
service_start_delay=5 # Delay in seconds before the service starts.

echo "[Mode: $mode]";

# Sudo check
if [[ $(id -u) -ne 0 ]] ; then echo "Please run as root" ; exit 1 ; fi

# Check systemctl is installed
if [[ -z "$(whereis systemctl | sed 's/systemctl: //')"  ]]; then echo "systemctl is not installed"; exit 1; fi
# Check rsync is installed	
if [[ -z "$(whereis rsync | sed 's/rsync: //')"  ]]; then echo "rsync is not installed"; exit 1; fi

if [[ "$mode" == "install" || "$mode" == "remove" ]]; then
	# Check service is presented and remove if yes.
	if [[ "$(systemctl --type=service | grep $service)" ]]; then
		echo "['$service' service is presented. Remove it.]";
		systemctl stop $service;
		systemctl disable $service;
		rm $service_dir/turbo-fan.service
		systemctl daemon-reload
	fi
		
	# Remove old files
	echo "[Remove old data]";
	rm -rvf $target_dir;
fi;

if [[ "$mode" == "install" ]]
then
	echo "[Create directories]";
	mkdir -p $target_dir

	echo "[Copy new data]";
	rsync -av ./* $target_dir --exclude=".git/*"

	echo "[Create turbo-fan service]"
	cat << EOF > $service_dir/turbo-fan.service
[Unit]
Description = Enables turbo button
After=sysinit.target
StartLimitIntervalSec=$service_start_delay

[Service]
Type=simple
Restart=no
RemainAfterExit=yes
User=root
WorkingDirectory=$target_dir
ExecStart=/bin/bash $target_dir/service.sh
ExecStop=/bin/bash ./uninstall.sh

[Install]
WantedBy=multi-user.target
EOF
	chown -R root:root $target_dir

KERNELVERSION=$(uname -r)

    cat << EOF > service.sh
KERNELVERSION="$KERNELVERSION"
cd $target_dir

rm /dev/acer-gkbbl-0 /dev/acer-gkbbl-static-0 -f

if [ "\$(uname -r)" != "\$KERNELVERSION" ]; then
	make clean
    source install.sh
	sed -i "s/^KERNELVERSION.*/KERNELVERSION=\"\$(uname -r)\"/" service.sh
else
	rmmod acer_wmi
	rmmod facer
	modprobe wmi
	modprobe sparse-keymap
	modprobe video
	insmod src/facer.ko
fi
EOF

	#locking down service file for security
	chown -R root:root $target_dir
	chmod 744 $target_dir/service.sh

	systemctl daemon-reload
	systemctl start $service
	systemctl enable $service
fi

LarryTheMagicDragon, 2025-03-31:
Can you show the output of running make?

Todd0042, 2025-04-18:
~/acer-predator-turbo-and-rgb-keyboard-linux-module main
❯ sudo ./install.sh
make -C /lib/modules/6.14.2-2-cachyos/build M=/home/todd/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.14.2-2-cachyos/build'
make[2]: Entering directory '/home/todd/acer-predator-turbo-and-rgb-keyboard-linux-module'
  CC [M]  src/facer.o
src/facer.c:1181:40: warning: tentative definition of variable with internal linkage has incomplete non-array type 'struct platform_profile_handler' [-Wtentative-definition-incomplete-type]
 1181 | static struct platform_profile_handler platform_profile_handler;
      |                                        ^
src/facer.c:1181:15: note: forward declaration of 'struct platform_profile_handler'
 1181 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/facer.c:2637:27: error: incomplete definition of type 'struct platform_profile_handler'
 2637 |                 platform_profile_handler.profile_get =
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~^
src/facer.c:1181:15: note: forward declaration of 'struct platform_profile_handler'
 1181 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/facer.c:2639:27: error: incomplete definition of type 'struct platform_profile_handler'
 2639 |                 platform_profile_handler.profile_set =
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~^
src/facer.c:1181:15: note: forward declaration of 'struct platform_profile_handler'
 1181 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/facer.c:2643:28: error: incomplete definition of type 'struct platform_profile_handler'
 2643 |                         platform_profile_handler.choices);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^
src/facer.c:1181:15: note: forward declaration of 'struct platform_profile_handler'
 1181 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/facer.c:2645:28: error: incomplete definition of type 'struct platform_profile_handler'
 2645 |                         platform_profile_handler.choices);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^
src/facer.c:1181:15: note: forward declaration of 'struct platform_profile_handler'
 1181 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/facer.c:2647:28: error: incomplete definition of type 'struct platform_profile_handler'
 2647 |                         platform_profile_handler.choices);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^
src/facer.c:1181:15: note: forward declaration of 'struct platform_profile_handler'
 1181 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/facer.c:2649:28: error: incomplete definition of type 'struct platform_profile_handler'
 2649 |                         platform_profile_handler.choices);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^
src/facer.c:1181:15: note: forward declaration of 'struct platform_profile_handler'
 1181 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/facer.c:2651:28: error: incomplete definition of type 'struct platform_profile_handler'
 2651 |                         platform_profile_handler.choices);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^
src/facer.c:1181:15: note: forward declaration of 'struct platform_profile_handler'
 1181 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/facer.c:2653:60: error: too few arguments to function call, expected 4, have 1
 2653 |                 err = platform_profile_register(&platform_profile_handler);
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~                          ^
/usr/lib/modules/6.14.2-2-cachyos/build/include/linux/platform_profile.h:50:16: note: 'platform_profile_register' declared here
   50 | struct device *platform_profile_register(struct device *dev, const char *name,
      |                ^                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   51 |                                          void *drvdata,
      |                                          ~~~~~~~~~~~~~~
   52 |                                          const struct platform_profile_ops *ops);
      |                                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/facer.c:2745:27: error: too few arguments to function call, single argument 'dev' was not specified
 2745 |                 platform_profile_notify();
      |                 ~~~~~~~~~~~~~~~~~~~~~~~ ^
/usr/lib/modules/6.14.2-2-cachyos/build/include/linux/platform_profile.h:58:6: note: 'platform_profile_notify' declared here
   58 | void platform_profile_notify(struct device *dev);
      |      ^                       ~~~~~~~~~~~~~~~~~~
[... 18 lines cut ...]
1 warning and 12 errors generated.
make[4]: *** [/usr/lib/modules/6.14.2-2-cachyos/build/scripts/Makefile.build:207: src/facer.o] Error 1
make[3]: *** [/usr/lib/modules/6.14.2-2-cachyos/build/Makefile:2002: .] Error 2
make[2]: *** [/usr/lib/modules/6.14.2-2-cachyos/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/home/todd/acer-predator-turbo-and-rgb-keyboard-linux-module'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.14.2-2-cachyos/build'
[... cut ...]
[  493.575547] [UFW BLOCK] IN=wlan0 OUT= MAC=4c:49:6c:d1:0e:c5:48:8d:36:d8:83:dd:08:00 SRC=10.0.0.19 DST=10.0.0.139 LEN=358 TOS=0x00 PREC=0x00 TTL=64 ID=14314 DF PROTO=UDP SPT=50543 DPT=53455 LEN=338
[  493.575559] [UFW BLOCK] IN=wlan0 OUT= MAC=4c:49:6c:d1:0e:c5:48:8d:36:d8:83:dd:08:00 SRC=10.0.0.19 DST=10.0.0.139 LEN=358 TOS=0x00 PREC=0x00 TTL=64 ID=14315 DF PROTO=UDP SPT=50543 DPT=53455 LEN=338
[  493.589500] [UFW BLOCK] IN=wlan0 OUT= MAC= SRC=fe80:0000:0000:0000:ac74:e9ba:4e53:489c DST=ff02:0000:0000:0000:0000:0000:0001:0003 LEN=79 TC=0 HOPLIMIT=255 FLOWLBL=99847 PROTO=UDP SPT=5355 DPT=5355 LEN=39
[*] Done

~/acer-predator-turbo-and-rgb-keyboard-linux-module main
❯

LarryTheMagicDragon, 2025-04-18:
@Todd0042 Have you tried #226?

Todd0042, 2025-04-18:
Just tried it, that makes the lighting work again, thanks! Still no noticeable difference for turbo mode.

Todd0042, 2025-04-18:
Is there a way to save a profile that is the same for all zones instead of saving 4 profiles, one for each zone?
such as:
./facer_rgb.py -m 0 -z 1,2,3,4 -b 75 -cR 000 -cB 000 -cG 255 -save allgreen

LarryTheMagicDragon, 2025-04-18:
Personally I just make scripts to color my keyboard how I want, and then make keyboard shortcuts for those scripts.

Irtesaam, 2025-06-25:
> Personally I just make scripts to color my keyboard how I want, and then make keyboard shortcuts for those scripts.

Did you find a way to save it for all zones, without relying on custom scripts ?

LarryTheMagicDragon, 2025-06-25:
@Irtesaam No, my scripts serve me well enough that I don't really think about it.

Irtesaam, 2025-06-25:
@LarryTheMagicDragon Could you share how your scripts work exactly? I tried creating separate profiles for each zone and running them together using systemd, but it felt a bit messy and redundant.

LarryTheMagicDragon, 2025-06-25:
This is my script to turn the keyboard blue. the & at the end of each line makes it go slightly faster, and makes a noticeable improvement.
```
#!/bin/bash
RED=0
BLUE=255
GREEN=0
/opt/turbo-fan/facer_rgb.py -m 0 -z 1 -cR $RED -cG $GREEN -cB $BLUE &
/opt/turbo-fan/facer_rgb.py -m 0 -z 2 -cR $RED -cG $GREEN -cB $BLUE &
/opt/turbo-fan/facer_rgb.py -m 0 -z 3 -cR $RED -cG $GREEN -cB $BLUE &
/opt/turbo-fan/facer_rgb.py -m 0 -z 4 -cR $RED -cG $GREEN -cB $BLUE &
```
I normally just set a static color, so creating a setkeyboard_white, or setkeyboard_amber script is is fairly quick and easy. I have a series of custom shortcuts on my keyboard to run the scripts to select what color I want, for example super+f4 sets the keyboard to white.