JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #235: Stopped working on Predator PHN16-71

yashyadurai, 2025-04-20:
## Is this an issue or a reproduction result?  ✔️ OR ⁉️
## Please fill in the template 
|Feature|Value|
|---	|---	|
System Information
	Manufacturer: Acer
	Product Name: Predator PHN16-71
	Version: V1.16
	Serial Number: NHQLTSI0064230088C7600
	UUID: cfffdfb5-23d2-0649-aa77-74d4dd67a213
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: RPL
	Product Name: Compass_RTX
	Version: V1.16
	Serial Number: NBQLT110024230088C7600
	Asset Tag: Type2 - Board Asset Tag
	Features:

Number of CPU Fans  1
Number of GPU Fans  1 
Number of RGB Zones --> 4
RGB keyboard works? --> No
Turbo button turns on fans? -->`No` 
|Turbo button turn on LED? --> No 
|Turbo button activates overclock?-->  No 
|Distribution Specific Information| No LSB modules are available.
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	2025.1
Codename:	kali-rolling

|Kernel Version|`uname -a`|

It used to work with previous version of Kali but after updating the kali version it just stopped working.
This is the response i get while doing the final step of installing.

└─$ sudo ./install_service.sh                           
[Mode: install]
['turbo-fan' service is presented. Remove it.]
Removed '/etc/systemd/system/multi-user.target.wants/turbo-fan.service'.
[Remove old data]
removed '/opt/turbo-fan/Makefile'
removed '/opt/turbo-fan/push.sh'
removed '/opt/turbo-fan/PKGBUILD/makepkg.sh'
removed '/opt/turbo-fan/PKGBUILD/PKGBUILD'
removed '/opt/turbo-fan/PKGBUILD/Predator-Sense-systemd-git.install'
removed directory '/opt/turbo-fan/PKGBUILD'
removed '/opt/turbo-fan/install_service.sh'
removed '/opt/turbo-fan/install_openrc.sh'
removed '/opt/turbo-fan/uninstall.sh'
removed '/opt/turbo-fan/src/facer.c'
removed directory '/opt/turbo-fan/src'
removed '/opt/turbo-fan/facer_rgb.py'
removed '/opt/turbo-fan/uninstall_service.sh'
removed '/opt/turbo-fan/keyboard.webp'
removed '/opt/turbo-fan/refresh.sh'
removed '/opt/turbo-fan/dkms.conf'
removed '/opt/turbo-fan/keyboard.py'
removed '/opt/turbo-fan/install.sh'
removed '/opt/turbo-fan/README.md'
removed '/opt/turbo-fan/LICENSE'
removed directory '/opt/turbo-fan'
[Create directories]
[Copy new data]
sending incremental file list
LICENSE
Makefile
README.md
dkms.conf
facer_rgb.py
install.sh
install_openrc.sh
install_service.sh
keyboard.py
keyboard.webp
push.sh
refresh.sh
service.sh
uninstall.sh
uninstall_service.sh
PKGBUILD/
PKGBUILD/PKGBUILD
PKGBUILD/Predator-Sense-systemd-git.install
PKGBUILD/makepkg.sh
src/
src/facer.c

sent 2,681,104 bytes  received 393 bytes  5,362,994.00 bytes/sec
total size is 2,679,088  speedup is 1.00
[Create turbo-fan service]
Created symlink '/etc/systemd/system/multi-user.target.wants/turbo-fan.service' → '/etc/systemd/system/turbo-fan.service'.

But doesn't work

LarryTheMagicDragon, 2025-04-20:
I don't see your kernel version. have you tried #226?

yashyadurai, 2025-05-08:
my kernel version is 6.12.20-amd64