JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #236: driver does not work after a system update

20192292012, 2025-04-21:
It worked initially, but after a system update, the driver stopped responding. When reinstalling it, the following logs appear in the terminal:  

make -C /lib/modules/6.14.2-1-default/build M=/home/jao/.local/modules/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/src/linux-6.14.2-1-obj/x86_64/default'
make[2]: Entering directory '/home/jao/.local/modules/acer-predator-turbo-and-rgb-keyboard-linux-module'
  CC [M]  src/facer.o
src/facer.c: In function ‘acer_platform_profile_setup’:
src/facer.c:2637:41: error: invalid use of undefined type ‘struct platform_profile_handler’
 2637 |                 platform_profile_handler.profile_get =
      |                                         ^
src/facer.c:2639:41: error: invalid use of undefined type ‘struct platform_profile_handler’
 2639 |                 platform_profile_handler.profile_set =
      |                                         ^
src/facer.c:2643:49: error: invalid use of undefined type ‘struct platform_profile_handler’
 2643 |                         platform_profile_handler.choices);
      |                                                 ^
src/facer.c:2645:49: error: invalid use of undefined type ‘struct platform_profile_handler’
 2645 |                         platform_profile_handler.choices);
      |                                                 ^
src/facer.c:2647:49: error: invalid use of undefined type ‘struct platform_profile_handler’
 2647 |                         platform_profile_handler.choices);
      |                                                 ^
src/facer.c:2649:49: error: invalid use of undefined type ‘struct platform_profile_handler’
 2649 |                         platform_profile_handler.choices);
      |                                                 ^
src/facer.c:2651:49: error: invalid use of undefined type ‘struct platform_profile_handler’
 2651 |                         platform_profile_handler.choices);
      |                                                 ^
src/facer.c:2653:49: error: passing argument 1 of ‘platform_profile_register’ from incompatible pointer type [-Wincompatible-pointer-types]
 2653 |                 err = platform_profile_register(&platform_profile_handler);
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 struct platform_profile_handler *
In file included from src/facer.c:26:
/usr/src/linux-6.14.2-1/include/linux/platform_profile.h:50:57: note: expected ‘struct device *’ but argument is of type ‘struct platform_profile_handler *’
   50 | struct device *platform_profile_register(struct device *dev, const char *name,
      |                                          ~~~~~~~~~~~~~~~^~~
src/facer.c:2653:23: error: too few arguments to function ‘platform_profile_register’
 2653 |                 err = platform_profile_register(&platform_profile_handler);
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/linux-6.14.2-1/include/linux/platform_profile.h:50:16: note: declared here
   50 | struct device *platform_profile_register(struct device *dev, const char *name,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
src/facer.c: In function ‘acer_thermal_profile_change’:
src/facer.c:2745:17: error: too few arguments to function ‘platform_profile_notify’
 2745 |                 platform_profile_notify();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/usr/src/linux-6.14.2-1/include/linux/platform_profile.h:58:6: note: declared here
   58 | void platform_profile_notify(struct device *dev);
      |      ^~~~~~~~~~~~~~~~~~~~~~~
src/facer.c: In function ‘acer_platform_probe’:
src/facer.c:3349:17: error: too few arguments to function ‘platform_profile_remove’
 3349 |                 platform_profile_remove();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/usr/src/linux-6.14.2-1/include/linux/platform_profile.h:53:5: note: declared here
   53 | int platform_profile_remove(struct device *dev);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
src/facer.c: In function ‘acer_platform_remove’:
src/facer.c:3372:17: error: too few arguments to function ‘platform_profile_remove’
 3372 |                 platform_profile_remove();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/usr/src/linux-6.14.2-1/include/linux/platform_profile.h:53:5: note: declared here
   53 | int platform_profile_remove(struct device *dev);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
src/facer.c: At top level:
src/facer.c:1181:40: error: storage size of ‘platform_profile_handler’ isn’t known
 1181 | static struct platform_profile_handler platform_profile_handler;
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~
src/facer.c:1998:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
 1998 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/src/linux-6.14.2-1/scripts/Makefile.build:208: src/facer.o] Error 1
make[3]: *** [/usr/src/linux-6.14.2-1/Makefile:2017: .] Error 2
make[2]: *** [/usr/src/linux-6.14.2-1/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/home/jao/.local/modules/acer-predator-turbo-and-rgb-keyboard-linux-module'
make[1]: *** [../../../linux-6.14.2-1/Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-6.14.2-1-obj/x86_64/default'
make: *** [Makefile:18: default] Error 2
rmmod: ERROR: Module acer_wmi is not currently loaded
rmmod: ERROR: Module facer is not currently loaded
insmod: ERROR: could not load module src/facer.ko: No such file or directory
[   11.530441] [    T574] wlp0s20f3: RX AssocResp from 50:91:e3:3e:0e:d5 (capab=0x1111 status=0 aid=19)
[   11.541893] [    T574] wlp0s20f3: associated
[   11.598681] [    T141] wlp0s20f3: Limiting TX power to 30 (30 - 0) dBm as advertised by 50:91:e3:3e:0e:d5
[   12.371455] [   T1888] Bluetooth: RFCOMM TTY layer initialized
[   12.371464] [   T1888] Bluetooth: RFCOMM socket layer initialized
[   12.371467] [   T1888] Bluetooth: RFCOMM ver 1.11
[   52.521222] [   T2300] JFS: nTxBlock = 8192, nTxLock = 65536
[   52.535623] [   T2326] NILFS version 2 loaded
[   52.673158] [   T2334] SGI XFS with ACLs, security attributes, realtime, quota, no debug enabled
[  129.333436] [   T2848] acer_wmi: Acer Laptop WMI Extras unloaded
[*] Done

it is a PT314-51 model with OpenSuse Tumbleweed, Linux 6.14.2-1-default

LarryTheMagicDragon, 2025-04-21:
#226 Should fix this. I've been a little too busy to make the patch compatible for all kernel versions.

20192292012, 2025-04-22:
its works. thank you

closed 2025-04-22 by 20192292012