JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #239: MAKEPKG ISSUE

zkhribach, 2025-05-08:
why does cp in PKGBUILD file have `--exclude ` option please fix this

here is what im getting

```bash
./makepkg.sh

[*] Build package script

[+] Creating Build directory
   ==> copying files to the building directory
   ==> go to building directory

[+] Building Package...

==> Making package: acer-predator-turbo-rgb-git 0.20241016-1 (Thu 08 May 2025 06:51:46 PM +01)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating acer-predator-turbo-and-rgb-keyboard-linux-module git repo...
==> Validating source files with sha256sums...
    acer-predator-turbo-and-rgb-keyboard-linux-module ... Skipped
==> Extracting sources...
  -> Creating working copy of acer-predator-turbo-and-rgb-keyboard-linux-module git repo...
Reset branch 'makepkg'
==> Starting pkgver()...
==> Removing existing $pkgdir/ directory...
==> Entering fakeroot environment...
==> Starting package()...
cp: unrecognized option '--exclude=.git'
Try 'cp --help' for more information.
==> ERROR: A failure occurred in package().
    Aborting...

[✘] build failed 
```

zkhribach, 2025-05-08:
@mmsaeed509 Can you fix this?

mmsaeed509, 2025-05-08:
- In this line

```bash
 cp -r ./* "${pkgdir}/opt/turbo-fan/" --exclude=".git"
``` 
- replace `cp -r ` with `rsync -av`

like : 

```bash
rsync -av ./* "${pkgdir}/opt/turbo-fan/" --exclude=".git"
```

I fixed it, pull the new changes and try again, it'll work Inshallah

closed 2025-05-08 by mmsaeed509