JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #243: PH18-71 Install worked, but no control!

ihaveanissueguys, 2025-05-11:
[sudo@archlinux ~]$ sudo dmidecode | grep "Product Name" -B 2 -A 4
[sudo] password for sudo: 
System Information
	Manufacturer: Acer
	Product Name: Predator PH18-71
	Version: V1.16
	Serial Number: NHQMNAA0013161E6B47600
	UUID: 85dd4517-77ea-604c-a58e-c01850fce5a4
	Wake-up Type: Power Switch
--
Base Board Information
	Manufacturer: RPL
	Product Name: EQS_RTX
	Version: V1.16
	Serial Number: NBQKS110013161E6B47600
	Asset Tag: Type2 - Board Asset Tag
	Features:
[sudo@archlinux ~]$ 

[sudo@archlinux ~]$ sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.aml
[sudo@archlinux ~]$ 

I was able to install the entire thing, however the python script did not do anything.

LarryTheMagicDragon, 2025-05-11:
Can you show the output of running install.sh?

ihaveanissueguys, 2025-05-11:
[sudo@archlinux acer-predator-turbo-and-rgb-keyboard-linux-module]$ sudo ./install.sh
[sudo] password for sudo: 
make -C /lib/modules/6.14.4-arch1-2/build M=/home/sudo/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.14.4-arch1-2/build'
make[2]: Entering directory '/home/sudo/acer-predator-turbo-and-rgb-keyboard-linux-module'
make[2]: Leaving directory '/home/sudo/acer-predator-turbo-and-rgb-keyboard-linux-module'
make[1]: Leaving directory '/usr/lib/modules/6.14.4-arch1-2/build'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
	sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
rmmod: ERROR: Module acer_wmi is not currently loaded
[   24.867230] wlan0: send auth to 10:78:5b:f5:e5:82 (try 1/3)
[   24.898756] wlan0: authenticated
[   24.909421] wlan0: associate with 10:78:5b:f5:e5:82 (try 1/3)
[   24.914023] wlan0: RX AssocResp from 10:78:5b:f5:e5:82 (capab=0x1411 status=0 aid=3)
[   24.925458] wlan0: associated
[   24.999497] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by 10:78:5b:f5:e5:82
[   42.656098] facer: Acer Laptop WMI Extras unloaded
[   42.693904] facer: Acer Laptop ACPI-WMI Extras
[   42.693929] facer: Function bitmap for Communication Button: 0x801
[   42.708483] input: Acer WMI hotkeys as /devices/virtual/input/input25
[*] Done
[sudo@archlinux acer-predator-turbo-and-rgb-keyboard-linux-module]$ 

[image]

LarryTheMagicDragon, 2025-05-11:
This all looks normal. can you tell me if anything works?

LarryTheMagicDragon, 2025-05-11:
@ihaveanissueguys I've slept on this, and I'd like to ask a follow up question. After you install with install.sh, what works and what doesn't work?

LarryTheMagicDragon, 2025-05-11:
@ihaveanissueguys Never mind on my previous question. I looked up your model and it looks like it has per key RGB, and this is only compatible with the older four zone RGB laptops.

ihaveanissueguys, 2025-05-22:
sad

closed 2025-05-22 by ihaveanissueguys