JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #248: Won't compile in 6.15.2

davidwkr, 2025-06-21:
`make -C /lib/modules/6.15.2-2-cachyos/build M=/home/latte/apps/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.15.2-2-cachyos/build'
make[2]: Entering directory '/home/latte/apps/acer-predator-turbo-and-rgb-keyboard-linux-module'
  CC [M]  src/facer.o
src/facer.c: In function ‘acer_platform_profile_setup’:
src/facer.c:2637:41: error: invalid use of undefined type ‘struct platform_profile_handler’
 2637 |                 platform_profile_handler.profile_get =
      |                                         ^
src/facer.c:2639:41: error: invalid use of undefined type ‘struct platform_profile_handler’
 2639 |                 platform_profile_handler.profile_set =
      |                                         ^
src/facer.c:2643:49: error: invalid use of undefined type ‘struct platform_profile_handler’
 2643 |                         platform_profile_handler.choices);
      |                                                 ^
src/facer.c:2645:49: error: invalid use of undefined type ‘struct platform_profile_handler’
 2645 |                         platform_profile_handler.choices);
      |                                                 ^
src/facer.c:2647:49: error: invalid use of undefined type ‘struct platform_profile_handler’
 2647 |                         platform_profile_handler.choices);
      |                                                 ^
src/facer.c:2649:49: error: invalid use of undefined type ‘struct platform_profile_handler’
 2649 |                         platform_profile_handler.choices);
      |                                                 ^
src/facer.c:2651:49: error: invalid use of undefined type ‘struct platform_profile_handler’
 2651 |                         platform_profile_handler.choices);
      |                                                 ^
src/facer.c:2653:49: error: passing argument 1 of ‘platform_profile_register’ from incompatible pointer type [-Wincompatible-pointer-types]
 2653 |                 err = platform_profile_register(&platform_profile_handler);
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 struct platform_profile_handler *
In file included from src/facer.c:26:
/usr/lib/modules/6.15.2-2-cachyos/build/include/linux/platform_profile.h:50:57: note: expected ‘struct device *’ but argument is of type ‘struct platform_profile_handler *’
   50 | struct device *platform_profile_register(struct device *dev, const char *name,
      |                                          ~~~~~~~~~~~~~~~^~~
src/facer.c:2653:23: error: too few arguments to function ‘platform_profile_register’; expected 4, have 1
 2653 |                 err = platform_profile_register(&platform_profile_handler);
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.15.2-2-cachyos/build/include/linux/platform_profile.h:50:16: note: declared here
   50 | struct device *platform_profile_register(struct device *dev, const char *name,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
src/facer.c: In function ‘acer_thermal_profile_change’:
src/facer.c:2745:17: error: too few arguments to function ‘platform_profile_notify’; expected 1, have 0
 2745 |                 platform_profile_notify();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.15.2-2-cachyos/build/include/linux/platform_profile.h:58:6: note: declared here
   58 | void platform_profile_notify(struct device *dev);
      |      ^~~~~~~~~~~~~~~~~~~~~~~
src/facer.c: In function ‘acer_platform_probe’:
src/facer.c:3349:17: error: too few arguments to function ‘platform_profile_remove’; expected 1, have 0
 3349 |                 platform_profile_remove();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.15.2-2-cachyos/build/include/linux/platform_profile.h:53:6: note: declared here
   53 | void platform_profile_remove(struct device *dev);
      |      ^~~~~~~~~~~~~~~~~~~~~~~
src/facer.c: In function ‘acer_platform_remove’:
src/facer.c:3372:17: error: too few arguments to function ‘platform_profile_remove’; expected 1, have 0
 3372 |                 platform_profile_remove();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.15.2-2-cachyos/build/include/linux/platform_profile.h:53:6: note: declared here
   53 | void platform_profile_remove(struct device *dev);
      |      ^~~~~~~~~~~~~~~~~~~~~~~
src/facer.c: At top level:
src/facer.c:1181:40: error: storage size of ‘platform_profile_handler’ isn’t known
 1181 | static struct platform_profile_handler platform_profile_handler;
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~
src/facer.c:1998:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
 1998 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/lib/modules/6.15.2-2-cachyos/build/scripts/Makefile.build:212: src/facer.o] Error 1
make[3]: *** [/usr/lib/modules/6.15.2-2-cachyos/build/Makefile:2039: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.2-2-cachyos/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/home/latte/apps/acer-predator-turbo-and-rgb-keyboard-linux-module'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.2-2-cachyos/build'
make: *** [Makefile:18: default] Error 2
rmmod: ERROR: Module acer_wmi is not currently loaded
rmmod: ERROR: Module facer is not currently loaded
insmod: ERROR: could not insert module src/facer.ko: Invalid module format
`

LarryTheMagicDragon, 2025-06-21:
@davidwkr I'm currently running 6.15.2-arch1-1, and I just downloaded and was able to build with make. I see `src/facer.c:2653:49: error: passing argument 1 of ‘platform_profile_register’ from incompatible pointer type [-Wincompatible-pointer-types] 2653 |                 err = platform_profile_register(&platform_profile_handler);`  in your output. which means line 2653 in your copy of the project contains `err = platform_profile_register(&platform_profile_handler);` whereas I looked in the copy I just downloaded, and line 2653 is currently just a `}`. Can you download to project again and tell us what happens when you try to build the latest?

closed 2025-06-23 by davidwkr

davidwkr, 2025-07-13:
@LarryTheMagicDragon I don't want to open another issue just in case it is just me:

❯ sudo ./install.sh
make -C /lib/modules/6.15.6-2-cachyos/build M=/home/latte/apps/acer-predator-turbo-and-rgb-keyboard-linux-module modules
make[1]: Entering directory '/usr/lib/modules/6.15.6-2-cachyos/build'
make[2]: Entering directory '/home/latte/apps/acer-predator-turbo-and-rgb-keyboard-linux-module'
  CC [M]  src/facer.o
src/facer.c:2083:20: warning: ‘WMID_gaming_get_u64’ defined but not used [-Wunused-function]
 2083 | static acpi_status WMID_gaming_get_u64(u64 *value, u32 cap)
      |                    ^~~~~~~~~~~~~~~~~~~
  MODPOST Module.symvers
  CC [M]  src/facer.mod.o
  CC [M]  .module-common.o
  LD [M]  src/facer.ko
  BTF [M] src/facer.ko
make[2]: Leaving directory '/home/latte/apps/acer-predator-turbo-and-rgb-keyboard-linux-module'
make[1]: Leaving directory '/usr/lib/modules/6.15.6-2-cachyos/build'
if [ -f "/var/lib/shim-signed/mok/MOK.priv" ] && [ -f "/var/lib/shim-signed/mok/MOK.der" ]; then \
        sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der src/facer.ko; \
fi
rmmod: ERROR: Module acer_wmi is not currently loaded
rmmod: ERROR: Module facer is not currently loaded
[   26.481282] wlan0: send auth to dc:36:0c:ac:e8:87 (try 1/3)
[   26.509441] wlan0: authenticated
[   26.509825] fs-verity: sha256 using implementation "sha256-ni"
[   26.510525] wlan0: associate with dc:36:0c:ac:e8:87 (try 1/3)
[   26.513223] wlan0: RX AssocResp from dc:36:0c:ac:e8:87 (capab=0x1411 status=0 aid=2)
[   26.518858] wlan0: associated
[   26.592927] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by dc:36:0c:ac:e8:87
[   37.508562] facer: Acer Laptop ACPI-WMI Extras
[   37.508595] facer: Function bitmap for Communication Button: 0x801
[   37.516909] input: Acer WMI hotkeys as /devices/virtual/input/input30
[*] Done

LarryTheMagicDragon, 2025-07-13:
@davidwkr I'm getting the same errors on my end. Thanks for reaching out, I'll start working on it.

LarryTheMagicDragon, 2025-08-30:
@davidwkr Sorry I haven't been able to get to this issue sooner. I see `[ 2921.866245] input: Acer WMI hotkeys as /devices/virtual/input/input53` in the output you listed, and as far as I am aware this means that the module successfully loaded and should work. On arch I booted into 6.12.44-1-lts, and I didn't find any issues. I know that's a bit newer than 6.12.36-2-cachyos-lts at this point, but I'm hoping you can give more description of what is wrong, as I can't tell what could be an issue from the logs.

LarryTheMagicDragon, 2025-08-31:
@davidwkr `[  106.619569] facer: Acer Laptop ACPI-WMI Extras` and `[  106.619602] facer: Function bitmap for Communication Button: 0x801` are printed from the module itself, so we know it's been compiled and loaded. Everything I see in the logs indicates it should be working. Can you try running the facer_rgb.py script and tell me what happens? Any details you can give about what has stopped working would be helpful.

davidwkr, 2025-09-16:
@LarryTheMagicDragon sorry! It worked after I updated. I forgot to say that when I posted the output again after updating it was working. I have no idea what happened the previous time if you say it should have worked. Anyways I'm happy I can use it :)

LarryTheMagicDragon, 2025-09-16:
@davidwkr I use this module myself, and I'm willing to pull out all the stops to get it working again when it fails. If you ever have any more issues just let me know.

LarryTheMagicDragon, 2026-03-22:
@davidwkr I'm on arch and I cannot recreate with either linux or linux-lts kernels right now. I'll look into setting up a cachyos build environment of some kind tomorrow, and see what I can do to fix it. I'm a bit busier these days, so if you don't hear anything within a couple of days don't hesitate to ask for an update.