JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module #259: PHN14-51

Haiado, 2025-08-15:
RGB works great, but Turbo does not. Pressing Turbo button does not do anything.

JafarAkhondali, 2025-08-16:
After pressing the turbo button, what is the output of:
```bash sudo dmesg | grep "Unknown function number"```

Lefrunila, 2025-10-21:
Having a completely different experience on PHN14-51 compared to @Haiado:
System Info:

Model: Predator PHN14-51
Kernel: 6.11.0-9-generic 
OS: Ubuntu 25.04 (Plucky)
BIOS: V1.01

What Happened:
Attempted one-time installation using ./install.sh. Unlike @Haiado's experience where RGB worked fine, I got an immediate kernel panic with blinking Caps Lock and complete system freeze. Had to hard restart.
RGB WMI Device Status:
```bash
/sys/bus/wmi/devices/7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56/ exists

**Last Kernel Messages Before Crash:**
Oct 20 22:25:23 kernel: facer: Acer Laptop ACPI-WMI Extras
Oct 20 22:25:23 kernel: facer: Function bitmap for Communication Button: 0x801
[System immediately froze - no further logs]
```

The module loaded successfully but crashed during/immediately after hardware initialization. No RGB functionality was tested as the system froze instantly.
Question for @Haiado: Can you share your kernel version, Ubuntu version, and BIOS version? The dramatic difference in behavior suggests:

Possible BIOS update changed WMI methods
Kernel 6.11.x (Ubuntu 25.04 dev) compatibility issue
Hardware revision differences in PHN14-51 units